gepd/Deviot #72: Can't select serial port

wuthula, 2016-09-26:
Hi I'm using sublime-text 3126 and can't get serial port to setup.
These are error from my console. Is there anything I can do to fix it??

Command ""/Users/iAN/Library/Application Support/Sublime Text 3/Packages/User/Deviot/penv/bin/python" -u -c "import setuptools, tokenize;__file__='/private/var/folders/02/lrycd0rd20b_zp_tqd5n5mf00000gn/T/pip-build-nxQ0rj/netifaces/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /var/folders/02/lrycd0rd20b_zp_tqd5n5mf00000gn/T/pip-gVJbc7-record/install-record.txt --single-version-externally-managed --compile --install-headers "/Users/iAN/Library/Application Support/Sublime Text 3/Packages/User/Deviot/penv/include/site/python2.7/netifaces"" failed with error code 1 in /private/var/folders/02/lrycd0rd20b_zp_tqd5n5mf00000gn/T/pip-build-nxQ0rj/netifaces/

Traceback (most recent call last):
  File "/Users/iAN/Library/Application Support/Sublime Text 3/Packages/Deviot/libs/mDNS.py", line 3, in <module>
    from zeroconf import ServiceBrowser, Zeroconf
ImportError: No module named zeroconf

monkianer, 2016-09-26:
I got it solved. Since this problem is after updated to macOS Sierra I need to agree to the Xcode/iOS license again.

gepd, 2016-09-26:
Thanks for the report, I'm currently working to fix some problems related with zeroconf and the option to select a serial port

closed 2016-10-18 by gepd

tablatronix, 2016-12-10:
Also having this issue.

I also get this, not sure if it has any relation
```
   xcrun: error: invalid active developer path (/Library/Developer/CommandLineTools), missing xcrun at: /Library/Developer/CommandLineTools/usr/bin/xcrun
    error: command 'cc' failed with exit status 1
```

tablatronix, 2016-12-10:
Any clue what 'agree to the xcode license" has to do with fixing this issue, or how ?

I also saw that the pip folder was being created and cleaned up, so maybe an order problem, not sure why wheel is failing.

what version of netifaces ? 
http://stackoverflow.com/a/28797566/1824214

gepd, 2016-12-11:
I really don't know what is the problem with xcode and since I don't have any mac machine I can't test it.

The last version of netifaces `0.10.5` has a problem so I'm installing `netifaces 0.10.4` 

You are right that path is wrong, I think it should be `Deviot/penv/Lib/site-packages` where is the call to that path?

tablatronix, 2016-12-11:
I ran out of time debugging was just trying to log what I was seeing. Ill dig into the source and see what this is trying to do when I have time. I think even master was failing. I gave up and had to do a restore to get back up.

tablatronix, 2016-12-28:
missing xcode command line tools ughh, must have been the osx upgrade

`xcode-select --install`
fixes for me

neophytos-theodorou, 2017-04-12:
Had a similar problem on Mint Linux. The workaround is to download the zeroconf.py from [https://github.com/paulsm/pyzeroconf/blob/master/Zeroconf.py](url) and save it in  ~/.config/sublime-text-3/Packages/Deviot/libs.
@gepd please add the specific file in the package distribution.