gepd/Deviot #73: save problem?

Encarnacao, 2016-09-27:
when i  used deviot with sublime something is always happening
lets imagen i start a new project and this will give me a name file of 147934605.ino
if i say save has and then put a new name and location where to save at, then everytime i will try to build file it always will give me error on build, is this some bug or is something im missing?

gepd, 2016-09-27:
It souldn't be a problem.

Please go to `Option > Full Verbose Output When Build`

Follow the steps until you get the error and paste here the output of the console, it will help me to know what is the problem, I tried to reproduced but I didn't the same result.

What is the version of Deviot you have installed?

Encarnacao, 2016-09-27:
[ Deviot 1.2.5 ] test.ino
03:22:54 Building the project | Processing...
[09/27/16 03:23:02] Processing uno (upload_protocol: stk500v2, platform: atmelavr, upload_flags: -Pusb, board: uno, framework: arduino)

avr-g++ -o "C:\Users\Asus\Desktop\test.ino.cpp" -x c++ -fpreprocessed -dD -E "c:\users\asus\appdata\local\temp\tmpg28w7i"
Collected 25 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <GSM> v1.0.6 (C:\Users\Asus\.platformio\packages\framework-arduinoavr\libraries\GSM)
|-- <Temboo> v1.1.4 (C:\Users\Asus\.platformio\packages\framework-arduinoavr\libraries\Temboo)
|   |-- <Bridge> v1.6.1 (C:\Users\Asus\.platformio\packages\framework-arduinoavr\libraries\Bridge)
|-- <EEPROM> v2.0 (C:\Users\Asus\.platformio\packages\framework-arduinoavr\libraries\EEPROM)
|-- <Wire> v1.0 (C:\Users\Asus\.platformio\packages\framework-arduinoavr\libraries\Wire)
|-- <HID> v1.0 (C:\Users\Asus\.platformio\packages\framework-arduinoavr\libraries\HID)
|-- <SoftwareSerial> v1.0 (C:\Users\Asus\.platformio\packages\framework-arduinoavr\libraries\SoftwareSerial)
|-- <Servo> v1.1.2 (C:\Users\Asus\.platformio\packages\framework-arduinoavr\libraries\Servo)
|-- <Firmata> v2.5.2 (C:\Users\Asus\.platformio\packages\framework-arduinoavr\libraries\Firmata)
|   |-- <SoftwareSerial> v1.0 (C:\Users\Asus\.platformio\packages\framework-arduinoavr\libraries\SoftwareSerial)
|   |-- <WiFi> v1.2.7 (C:\Users\Asus\.platformio\packages\framework-arduinoavr\libraries\WiFi)
|   |   |-- <SPI> v1.0 (C:\Users\Asus\.platformio\packages\framework-arduinoavr\libraries\SPI)
|-- <Bridge> v1.6.1 (C:\Users\Asus\.platformio\packages\framework-arduinoavr\libraries\Bridge)
|-- <WiFi> v1.2.7 (C:\Users\Asus\.platformio\packages\framework-arduinoavr\libraries\WiFi)
|   |-- <SPI> v1.0 (C:\Users\Asus\.platformio\packages\framework-arduinoavr\libraries\SPI)
|-- <LiquidCrystal> v1.0.5 (C:\Users\Asus\.platformio\packages\framework-arduinoavr\libraries\LiquidCrystal)
|-- <SD> v1.0.8 (C:\Users\Asus\.platformio\packages\framework-arduinoavr\libraries\SD)
|   |-- <SPI> v1.0 (C:\Users\Asus\.platformio\packages\framework-arduinoavr\libraries\SPI)
|-- <Ethernet> v1.1.2 (C:\Users\Asus\.platformio\packages\framework-arduinoavr\libraries\Ethernet)
|   |-- <SPI> v1.0 (C:\Users\Asus\.platformio\packages\framework-arduinoavr\libraries\SPI)
|-- <Robot Control> v1.0.2 (C:\Users\Asus\.platformio\packages\framework-arduinoavr\libraries\Robot_Control)
|   |-- <SPI> v1.0 (C:\Users\Asus\.platformio\packages\framework-arduinoavr\libraries\SPI)
|   |-- <Wire> v1.0 (C:\Users\Asus\.platformio\packages\framework-arduinoavr\libraries\Wire)
|-- <SPI> v1.0 (C:\Users\Asus\.platformio\packages\framework-arduinoavr\libraries\SPI)

*** Multiple ways to build the same target were specified for: C:\Users\Asus\AppData\Local\Temp\Deviot\Desktop\.pioenvs\uno\src\Arduino_Sketches\arduino-1.6.5-r5\hardware\arduino\avr\cores\arduino\wiring_pulse.o  (from ['C:\\Users\\Asus\\Desktop\\Arduino_Sketches\\arduino-1.6.5-r5\\hardware\\arduino\\avr\\cores\\arduino\\wiring_pulse.S'] and from ['C:\\Users\\Asus\\Desktop\\Arduino_Sketches\\arduino-1.6.5-r5\\hardware\\arduino\\avr\\cores\\arduino\\wiring_pulse.c'])
File "C:\Users\Asus\AppData\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 98, in BuildProgram
 [ERROR] Took 44.16 seconds 
03:23:46 ERROR | it took 51.42s

gepd, 2016-09-27:
This line `*** Multiple ways to build the same target were specified for: ` Tell me error is in the `platformio.ini` file. But I couldn't get reproduce the same issue, so let's go step by step.

First, remove All files in `C:\Users\Asus\AppData\Local\Temp\Deviot`

As I understand the steps you follow to get the issue are:

1. Create an sketch:
````c++
void setup()
{
    
}
void loop()
{
    
}
```

2. Build it to Deviot assign a name (like `186934101`)
3. The file with other name and path from `ST Menu > File > Save As` Here you save the file with the name `newsketchname.ino` in `desktop/newfolder`

4. You build the file

Now is when you get the error posted above?

Encarnacao, 2016-09-27:
hard to explain what is happening, for exemple right now if i click new sketch will ask me for a path so lets say i select C:\Users\Asus\AppData\Local\Temp\Deviot
after when i click build the file will just disapear
no idea what is going on

gepd, 2016-09-27:
Please use the [develop](https://github.com/gepd/Deviot/tree/develop) branch and tell me if you get the same issue. The issue where the file just disapear I've fix some days ago, the first one I can't reproduce it yet.

To install the develop branch just replace the files in `Packages/Deviot` by the files in the `zip`

gepd, 2017-06-30:
Fixed in Deviot 2.0

closed 2017-06-30 by gepd