gepd/Deviot #74: Where is Deviot's platformio located?

garvamel, 2016-09-27:
I need to add a custom board. Thanks!

gepd, 2016-09-27:
Follow this topic, specially for the json format: https://github.com/gepd/Deviot/issues/58 (last comment)

At this moment I haven't any option go directly to the boards folder, but and easy way is:

`Deviot > Library Options > Open Library Folder` You should back one directory until `.platformio`. Create a new folder `boards` and save your json file there.

To update the board list, look https://github.com/gepd/Deviot/issues/58#issuecomment-244584498

garvamel, 2016-09-27:
Thanks!

As a follow up question, can I point Deviot to the system-wide installation of Platformio? It seems a little bit more intuitive and useful to me.

closed 2016-09-29 by gepd