gepd/Deviot #75: Upload fails for NodeMCU v1.0 on Windows

Suyash458, 2016-10-05:
Hi, I'm trying to upload some code to a NodeMCU but it fails each time saying 
"Do not know how to make File target `program' " (Verbose output enabled)

Building the code succeeds

gepd, 2016-10-05:
Please, paste here the full console output so I can see where is the problem

Suyash458, 2016-10-06:
Sure, here you go. 
```
[ Deviot 1.2.5 ] MPU9150BasicAHRS.ino
02:15:59 Uploading firmware | [10/06/16 02:16:03] Processing nodemcuv2 (upload_protocol: stk500v2, platform: espressif8266, board: nodemcuv2, framework: arduino)

xtensa-lx106-elf-g++ -o "D:\Projects\Fitbit\Libraries and code\MPU-9150-master\MPU9150BasicAHRS\MPU9150BasicAHRS.ino.cpp" -x c++ -fpreprocessed -dD -E "c:\users\hal9000\appdata\local\temp\tmpfkelm0"
Collected 23 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <Wire> v1.0 (C:\Users\HAL9000\.platformio\packages\framework-arduinoespressif8266\libraries\Wire)
*** Do not know how to make File target `program' (C:\Users\HAL9000\AppData\Local\Temp\Deviot\MPU9150BasicAHRS\program).  Stop.
 [ERROR] Took 4.27 seconds 
02:16:07 ERROR | it took 8.20s
```

gepd, 2016-10-06:
Seems like  you are using the programmer option. 

Go to `Deviot > Programmer` and choose `None`

Suyash458, 2016-10-06:
I just copied whatever settings I was using on Arduino. 
Choosing None worked, thanks!

closed 2016-10-06 by Suyash458