gepd/Deviot #77: Custom board shows up twice

garvamel, 2016-10-11:
Added a custom board as instructed in #74, but the board appears twice in both `Select Board` and `Select Environment`.

Env:
macOS Sierra
ST 3
Deviot 1.2.5

Thanks!

gepd, 2016-10-11:
Please, upload the boardname.json so I can test it here

garvamel, 2016-10-12:
File is located at `~/.platformio/boards/` and nowhere else.

Thanks!

[4MHzatmega328p.json.zip](https://github.com/gepd/Deviot/files/523224/4MHzatmega328p.json.zip)

garvamel, 2016-10-12:
Culprit is `platformio_boards.json` in `User/Deviot` in sublime packages folder. It keeps the definition sotred even if files are removed. I had to remove deviot preference files for the board to be removed but after restart and adding the board again, it shows up twice once more

gepd, 2016-10-12:
Ok I found the problem, you didn't set the platform in the json file, adding it solve the problem:

````json
{
  "build": {
    "core": "arduino", 
    "extra_flags": "-DARDUINO_ARCH_AVR", 
    "f_cpu": "4000000L", 
    "mcu": "atmega328p", 
    "variant": "standard"
  }, 
  "frameworks": ["arduino"],
  "name": "Xenitt Gaviot (3.3V, 4MHz)",
  "platform": "atmelavr",
  "upload": {
    "maximum_ram_size": 2048, 
    "maximum_size": 32256, 
    "protocol": "arduino", 
    "require_upload_port": true, 
    "speed": 19200
  }, 
  "url": "http://www.xenitt.cl", 
  "vendor": "Xenitt"
}
```

garvamel, 2016-10-12:
Awesome, thanks!

closed 2016-10-12 by gepd