gepd/Deviot #78: upload via LAN ?

facsa79, 2016-10-19:
I can not pass avrdude option `-P net:remoteIP:Port` via platformio.ini
Situation: 
- in my LAN there is a raspberry PI connected to an arduinoMega2560 via USB
- on raspberry /dev/ttyACM0 (port of the mega) is redirected to tcpPort:50505 
with the following command: `socat tcp-l:50505,reuseaddr,fork \   file:/dev/ttyACM0,nonblock,raw,echo=0,waitlock=/var/run/ttyACM0.lock,b115200`
- on my pc (debian jessie) I can upload the sketch via LAN with `avrdude -P net:raspberryIP:50505` with avrdude directly and with eclipse. In eclipse it's possible to set the port to net:raspberryIP:50505 for upload and for the serial monitor

It would be nice a possibility in Deviot to upload and use the serial monitor via this way.

gepd, 2016-10-19:
Follow this steps:

1. Go to `Deviot Menu > Serial Ports > Add IP Manually`

2. Add `net:raspberryIP:50505`

And try to upload with it... I've not tested, but it should do it. 

If it shows an error go to `Deviot Menu > Options > Full Verbose Output When Build` 
Try to upload again and copy the console output here

facsa79, 2016-10-19:
console output:
Traceback (most recent call last):
  File "/home/facsa/.config/sublime-text-3/Packages/Deviot/DeviotStarter.py", line 596, in on_done
    Menu().createSerialPortsMenu()
AttributeError: 'Menu' object has no attribute 'createSerialPortsMenu'

gepd, 2016-10-23:
Sorry about the delay, can you try using the develop branch? 

just download [this](https://github.com/gepd/Deviot/archive/develop.zip) zip, extract it in `Packages/Deviot` and try my previous recomendation again

facsa79, 2016-10-23:
I've tried
It's almost fine.
Now It is possible to select the the required port, but maybe somewhere there is a checking method which try to find the selected port ( probably in /dev ), it give back an error before avrdude  could receive the parameters.
 ```
[...]
avr-objcopy -O ihex -R .eeprom .pioenvs/megaatmega2560/firmware.elf .pioenvs/megaatmega2560/firmware.hex
BeforeUpload(["upload"], [".pioenvs/megaatmega2560/firmware.hex"])
Use manually specified: net:192.168.1.40:50505
*** [upload] could not open port net:192.168.1.40:50505: [Errno 2] No such file or directory: 'net:192.168.1.40:50505'
========================== [ERROR] Took 0.85 seconds ==========================
11:05:27 ERROR | it took 0.98s
```

gepd, 2016-10-24:
It's seems to be a problem with Deviot, I was looking how to solve the problem but as I'm working in the version 2 of the plugin I don't think I should spend much time on it

I  propose you an alternative to upload using the build system of sublime text. 

First you need to install platformio in your machine use this commad to install it `sudo pip install -U platformio` ([full instructions](http://docs.platformio.org/en/stable/installation.html))

Follow [this](http://docs.platformio.org/en/stable/ide/sublimetext.html#manual-integration) instructions to add the commands in sublime text.

Finally to you need to edit the platformio.ino file (you can open ir from `Deviot Menu > Open Build Config File` ) you must have your sketch open and selected to this command work

The config file will be opened in a new window, add `upload_port = net:192.168.1.40:50505`

Your file should look like this

```
[env:megaatmega2560]
platform = atmelavr
board = megaatmega2560
framework = arduino
upload_port = net:192.168.1.40:50505
```

Now you should build the project (`cmd + b`) but before make sure you have selected the platformio option in `Tools > Build System`

I'm sorry if it's confusing, as I already wrote I'm working in the version 2 of Deviot and because all this kind of problem I decided to start to write the plugin from scratch and I still don't have an alpha version with the basic commands to work

Meanwhile I'll be here to help as much as I can

facsa79, 2016-10-24:
With the modified deviot version I can upload the sketch.
The 2nd problem was not in deviot, but in platformio.
`*** [upload] could not open port net:192.168.1.40:50505: [Errno 2] No such file or directory: 'net:192.168.1.40:50505'`
After some hours in analyzing the platformio code I realized the problem is with the serial flushing method of platformio before upload. For upload via net it must be disabled.
Finally, in the board file `~/.platformio/platforms/atmelavr/boards/megaatmega2560.json` under upload part I added  `"disable_flushing": true` It  skips the serial flushing before upload.
```
[...]
"upload": {
    "disable_flushing": true,
    "maximum_ram_size": 8192, 
[...]
```
With your modification and with this change in  board file the problem is solved.
Thank you for your help!

closed 2016-10-24 by facsa79

gepd, 2016-10-24:
Thanks for your report @facsa79 in the future, I'll add this solution to the wiki