gepd/Deviot #79: select custom board

hackermagnet, 2016-10-23:
Hello,
I have created a few custom boards but I cannot find them in Deviot.
I can see my boards if I execute ```platformio boards``` in terminal.
Should I update the list in Deviot somehow?

gepd, 2016-10-23:
I'm improving this feature in v2 of deviot, in this version you should remove the file located in: `Packages\User\Deviot\Preset\platformio_boards.json` and restart ST

It will took a few seconds to refresh and show the new board

hackermagnet, 2016-10-24:
I deleted this file but it didn't reappear after restarting ST. Also it wasn't showing the boards list to select one.
It reappeared after building a project, disabling and enabling Deviot.
I found that it changed the structure of my boards json files.
I had something like that:
```{
  "my_board_name":{
    "build": {}, 
    "frameworks": [], 
    "name": "", 
    "platform": "", 
    "upload": {}, 
    "url": "", 
    "vendor": ""
  }
}```

which turned into this:
```{
  "build": {}, 
  "frameworks": [], 
  "name": "", 
  "platform": "", 
  "upload": {}, 
  "url": "", 
  "vendor": ""
}```

After this change I can see my boards on Deviot but it gives me an error if I execute ```platformio boards``` in terminal.

gepd, 2016-10-24:
Your first structure is like this:? https://github.com/gepd/Deviot/issues/77#issuecomment-253287689

Try again, and this time after delete the file in the `Preset` folder, go to `Deviot > Language` and select your current language, it will ask to restart Deviot, accept it and tell me if the behavior is the same

hackermagnet, 2016-10-24:
No, the structure I was using is like [this](https://github.com/gepd/Deviot/issues/58#issuecomment-244584498) because I found that only with this structure I'm not getting errors on terminal.

After the structure change they are like the commment you posted but I get this error
```
Error: Traceback (most recent call last):
  File "/Library/Python/2.7/site-packages/platformio/__main__.py", line 107, in main
    cli(None, None, None)
  File "/Library/Python/2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Library/Python/2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Library/Python/2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
```

gepd, 2016-10-24:
it seems to be a problem with PlatformIO.

The last structure is working only with PlatformIO 3, having open a cpp or ino file, Deviot will show you the version installed in the status bar, if your version is 2.x I suggest you to update from

`Deviot Menu > Options > Upgrade PlatformIO` 

After install the last version (`3.1.0` in this moment) you shouldn't have problem but If you keep getting errors, you can remove the folder `Deviot` in `Packages\User` it will make to install PlatformIO again, this time all errors should be gone

And you should use [this](https://github.com/gepd/Deviot/issues/58#issuecomment-247852879) json structure for your board

hackermagnet, 2016-10-25:
I have Deviot 1.2.5 and PlatformIO 3.1.0
I use Mac OS X 10.12 and ST build 3126.
I tried all your suggestions with no luck.
I created just one board with the structure you suggested.
Everything is working fine on ST but still getting the same error in Terminal.