gepd/Deviot #80: Got"avrdude: ser_open(): can't open device "unknown": No such file or directory" while uploading to Chinese cheap version of Arduino Nano

simonliu009, 2016-10-24:
I already selected "Arduino Nano with ATmega328" in the "Board" and "Enviroment", and selected "dev/cu.wchusbserial1410" in serial port,  and "AVRISP MKII" as programmer (actually I tried every item in the programmer. But I got the following error:
Using Port                    : unknown
Using Programmer              : stk500v2
avrdude: ser_open(): can't open device "unknown": No such file or directory

I think the serial port is not recognized correctly. But how do I fix it? I've tried many times and got the same error message.

gepd, 2016-10-24:
Hi @simonliu009 I have been notified of this bug before, I'm currently developing the version 2 of the plugin where this bug should be solved.

If you let the programmer in `none` you shouldn't have problem to upload your sketch

simonliu009, 2016-10-25:
wow, I never thought of using "none". It is working now.

closed 2017-07-04 by gepd