gepd/Deviot #81: Progress Thread hangs on compile failure

ubidefeo, 2016-10-24:
hi there

I have been noticing this behaviour lately.
When a compile task fails, the bottom of the window keeps displaying a progress bar in which the `=` indicator keeps jumping to random positions.
I think the Progress Thread gets stuck and doesn't timeout or something.

here's a screenshot of a failed compilation

[image]

gepd, 2016-10-24:
Thanks for your report @ubidefeo I've noticed this behavior too. The plugin has a lot of this kind of bugs, that is why I started to write the plugin from scratch, I still don't have all the basic functions to release a  alpha, but I hope to do it soon.

gepd, 2017-06-20:
Please read https://github.com/gepd/Deviot/issues/117

closed 2017-06-26 by gepd