gepd/Deviot #82: deeper nested examples folders not usable

tablatronix, 2016-10-24:
Develop branch
Does not support nested examples folders beyond first level.

Install `blynk` library, goto examples, all example files are 2 levels deep in subfolders, but nothing happens when you try to drill down past first level.

gepd, 2016-10-26:
Thanks @tablatronix I've reproduced the bug, I'm working on Deviot V2, so I'm rewriting most of the code, I'll fix this issue in the new development. I'll let you know when a pre-release is ready to test

tablatronix, 2017-09-17:
deeper subfolders still do not work

eg.

blynk library
blynk->more->setproperty->2 subfolders here menu just closes

gepd, 2017-09-20:
Fixed again, now it will open the sketch in any level

closed 2017-09-21 by gepd

tablatronix, 2017-09-22:
Still having issue on OSX

tablatronix, 2017-09-22:
working