gepd/Deviot #83: Serial console buffer enhancements

tablatronix, 2016-10-28:
Sublime really chokes when your serial console ( or any open file ) has many many lines.

* Add capability to manually clear serial console/file just for convenience.
* Add capability for max lines auto clearing of serial console/file for large amounts of data 30,000 lines etc.

gepd, 2016-10-28:
I'm just working in the `command line` feature and this option is necessary there too. At this momment I think I've the manually clear option, max lines auto clearing could be possible too, I'll add this as millestione for v 2.0

tablatronix, 2016-10-29:
Yeah a simple manual clear menu option shortcut would be wonderful for the time being.

RebelDesigner, 2017-02-03:
Is this feature in development? I really need to be able to clear the Serial monitor screen.
Also, output to the serial monitor, or when compiling lots of files with lots of compiler output tends to be slow to output to the Deviot console. I know that reading such stuff can be a pain from other projects, but there are improvements possible. Love Deviot (and Sublime), best plugin imho. Great work.

gepd, 2017-06-18:
Already implemented. 

I'll release the developer version soon, I'll need testing the plugin

closed 2017-06-18 by gepd