gepd/Deviot #86: Some issues during the instalecion.

rv9yw, 2016-11-10:
The logs:

> 17:31:03 PlatformIO is not installed
> Preparing to installing it, please be patient it may take a while.
> 17:31:03 Extracting files...
> 17:31:03 Installing PlatformIO...
> 17:31:08 Error Creating the virtualenv

> > plugins loaded
> Could not import subprocess32 module, falling back to subprocess module
> ""D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\Scripts\pio"" ­Ґ пў«пҐвбп ў­гваҐ­­Ґ© Ё«Ё ў­ и­Ґ©
> Є®¬ ­¤®©, ЁбЇ®«­пҐ¬®© Їа®Ја ¬¬®© Ё«Ё Ї ЄҐв­л¬ д ©«®¬.
> New python executable in D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\Scripts\python.exe
> Installing setuptools, pip, wheel...
>   Complete output from command D:\arduino\soft\Subl...v\Scripts\python.exe -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel:
>   Collecting setuptools
> Exception:
> Traceback (most recent call last):
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\basecommand.py", line 209, in main
>     status = self.run(options, args)
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\commands\install.py", line 299, in run
>     requirement_set.prepare_files(finder)
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\req\req_set.py", line 359, in prepare_files
>     ignore_dependencies=self.ignore_dependencies))
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\req\req_set.py", line 511, in _prepare_file
>     finder, self.upgrade, require_hashes)
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\req\req_install.py", line 277, in populate_link
>     self.link = finder.find_requirement(self, upgrade)
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\index.py", line 436, in find_requirement
>     all_candidates = self.find_all_candidates(req.name)
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\index.py", line 394, in find_all_candidates
>     for page in self._get_pages(url_locations, project_name):
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\index.py", line 539, in _get_pages
>     page = self._get_page(location)
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\index.py", line 642, in _get_page
>     return HTMLPage.get_page(link, session=self.session)
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\index.py", line 751, in get_page
>     "Cache-Control": "max-age=600",
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\_vendor\requests\sessions.py", line 480, in get
>     return self.request('GET', url, **kwargs)
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\download.py", line 377, in request
>     return super(PipSession, self).request(method, url, *args, **kwargs)
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\_vendor\requests\sessions.py", line 468, in request
>     resp = self.send(prep, **send_kwargs)
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\_vendor\requests\sessions.py", line 608, in send
>     r.content
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\_vendor\requests\models.py", line 737, in content
>     self._content = bytes().join(self.iter_content(CONTENT_CHUNK_SIZE)) or bytes()
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\_vendor\requests\models.py", line 660, in generate
>     for chunk in self.raw.stream(chunk_size, decode_content=True):
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\_vendor\requests\packages\urllib3\response.py", line 344, in stream
>     data = self.read(amt=amt, decode_content=decode_content)
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\_vendor\requests\packages\urllib3\response.py", line 301, in read
>     data = self._fp.read(amt)
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\_vendor\cachecontrol\filewrapper.py", line 54, in read
>     self.__callback(self.__buf.getvalue())
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\_vendor\cachecontrol\controller.py", line 297, in cache_response
>     self.serializer.dumps(request, response, body=body),
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\download.py", line 280, in set
>     return super(SafeFileCache, self).set(*args, **kwargs)
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\_vendor\cachecontrol\caches\file_cache.py", line 99, in set
>     with self.lock_class(name) as lock:
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\_vendor\lockfile\mkdirlockfile.py", line 19, in __init__
>     LockBase.__init__(self, path, threaded, timeout)
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\virtualenv\virtualenv_support\pip-8.0.2-py2.py3-none-any.whl\pip\_vendor\lockfile\__init__.py", line 242, in __init__
>     hash(self.path)))
>   File "D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\lib\ntpath.py", line 85, in join
>     result_path = result_path + p_path
> UnicodeDecodeError: 'ascii' codec can't decode byte 0xcf in position 6: ordinal not in range(128)
> ----------------------------------------
> ...Installing setuptools, pip, wheel...done.
> Traceback (most recent call last):
>   File "virtualenv.py", line 2304, in <module>
>     main()
>   File "virtualenv.py", line 705, in main
>     symlink=options.symlink)
>   File "virtualenv.py", line 924, in create_environment
>     download=download,
>   File "virtualenv.py", line 877, in install_wheel
>     call_subprocess(cmd, show_stdout=False, extra_env=env)
>   File "virtualenv.py", line 783, in call_subprocess
>     % (cmd_desc, proc.returncode))
> OSError: Command D:\arduino\soft\Subl...v\Scripts\python.exe -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

Python, pip, virtualenv are alredy installed. 
Could you help me, please?

rv9yw, 2016-11-10:
Forgot! Windows 7 64bit, Sublime Text 3126.

rv9yw, 2016-11-10:
AFter I changed language of system I have another issue at the log.

```
New python executable in D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\Scripts\python.exe
Traceback (most recent call last):
  File "virtualenv.py", line 2304, in <module>
    main()
  File "virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "virtualenv.py", line 904, in create_environment
    site_packages=site_packages, clear=clear, symlink=symlink))
  File "virtualenv.py", line 1197, in install_python
    shutil.copyfile(executable, py_executable)
  File "D:\1_python\python_2_7\lib\shutil.py", line 83, in copyfile
    with open(dst, 'wb') as fdst:
IOError: [Errno 13] Permission denied: 'D:\\arduino\\soft\\Sublime\\Data\\Packages\\User\\Deviot\\penv\\Scripts\\python.exe'

Package Control: Error parsing JSON from channel https://github.com/gepd/Stino/tree/new-stino.
```

gepd, 2016-11-11:
This issue seems to be similar to: https://github.com/gepd/Deviot/issues/38

The problem is, until now I have not been able to replicate this issue.

Is your sublime portable?

I've tried a portable version in my machine but I don't get that error

rv9yw, 2016-11-11:
Yes, it's portable.
A bit later I'll try in another machine too.

rv9yw, 2016-11-13:
Hi. 
I didn't try in another machine.
However, I solved out the problem with cyrillic symbols. Deleted package Deviot, Deleted Package Stino, Deleted all folders which was created by Deviot. 
Restarted my PC and started  installation again and now it's lasting more than hour.
LOOK  at THIS - http://s8.hostingkartinok.com/uploads/images/2016/11/78b41eb9fa3c03478e6124424f629d05.png

gepd, 2016-11-13:
It's shouldn't take more than a few minutes. Please paste here the output of the ST console (`ST Menu > View > Show console` )

rv9yw, 2016-11-13:
> DPI scale: 1
startup, version: 3126 windows x64 channel: stable
executable: /D/arduino/soft/Sublime/sublime_text.exe
working dir: /D/arduino/soft/Sublime
packages path: /D/arduino/soft/Sublime/Data/Packages
state path: /D/arduino/soft/Sublime/Data/Local
zip path: /D/arduino/soft/Sublime/Packages
zip path: /D/arduino/soft/Sublime/Data/Installed Packages
ignored_packages: ["Deviot", "Vintage"]
pre session restore time: 0.170735
loading dictionary Packages/Language - English/en_US.dic
startup time: 0.241735
first paint time: 0.251735
first paint time: 0.251735
reloading plugin Default.auto_indent_tag
reloading plugin Default.block
reloading plugin Default.comment
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.install_package_control
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.quick_panel
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.settings
reloading plugin Default.show_scope_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin 0_package_control_loader.00-package_control
reloading plugin Package Control.1_reloader
reloading plugin Package Control.2_bootstrap
reloading plugin Package Control.Package Control
reloading plugin SideBarEnhancements.SideBar
reloading plugin SideBarEnhancements.SideBarAPI
reloading plugin SideBarEnhancements.SideBarDefaultDisable
reloading plugin SideBarEnhancements.Stats
reloading plugin SyncedSideBar.SyncedSideBar
reloading plugin SublimeCodeIntel.ordereddict
reloading plugin SublimeCodeIntel.SublimeCodeIntel
reloading plugin SublimePythonIDE.__init__
reloading plugin SublimePythonIDE.pep8
reloading plugin SublimePythonIDE.sublime_python
reloading plugin SublimePythonIDE.sublime_python_colors
reloading plugin SublimePythonIDE.sublime_python_commands
reloading plugin SublimePythonIDE.sublime_python_errors
reloading plugin SublimePythonIDE.sublime_python_linting
reloading plugin SublimePythonIDE.sublime_python_refactorings
reloading plugin SublimeREPL.__init__
reloading plugin SublimeREPL.completions
reloading plugin SublimeREPL.lang_integration
reloading plugin SublimeREPL.run_existing_command
reloading plugin SublimeREPL.sublimerepl
reloading plugin SublimeREPL.sublimerepl_build_system_hack
reloading plugin SublimeREPL.text_transfer
plugins loaded
Package Control: The package Deviot is being re-enabled after a Package Control operation was interrupted
ignored packages updated to: ["Vintage"]
Package Control: Skipping automatic upgrade, last run at 2016-11-13 09:57:36, next run at 2016-11-13 10:57:36 or after
reloading settings Packages/User/Package Control.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings
started server on port 53596 with ['D:\\1_python\\python_2_7\\python.exe']
WINDOW COMMAND ENABLED True
WINDOW COMMAND ENABLED True
WINDOW COMMAND ENABLED True
WINDOW COMMAND ENABLED True
WINDOW COMMAND ENABLED True
WINDOW COMMAND ENABLED False
Package Control: Error parsing JSON from channel https://github.com/gepd/Stino/tree/new-stino.
ignored packages updated to: ["Deviot", "Vintage"]
reloading settings Packages/User/Package Control.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings
Package Control: Error parsing JSON from channel https://github.com/gepd/Stino/tree/new-stino.
reloading settings Packages/User/Package Control.sublime-settings
ignored packages updated to: ["Vintage"]
reloading plugin Deviot.DeviotStarter
Could not import subprocess32 module, falling back to subprocess module
generating syntax summary
generating meta info summary
""D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\Scripts\pio"" ­Ґ пў«пҐвбп ў­гваҐ­­Ґ© Ё«Ё ў­Ґи­Ґ©
Є®¬ ­¤®©, ЁбЇ®«­пҐ¬®© Їа®Ја ¬¬®© Ё«Ё Ї ЄҐв­л¬ д ©«®¬.

> reloading settings Packages/User/Package Control.sublime-settings
> reloading settings Packages/User/Preferences.sublime-settings
> WINDOW COMMAND ENABLED True
> WINDOW COMMAND ENABLED True
> WINDOW COMMAND ENABLED True
> WINDOW COMMAND ENABLED True
> WINDOW COMMAND ENABLED True
> WINDOW COMMAND ENABLED False
> started server on port 60154 with ['D:\\1_python\\python_2_7\\python.exe']

rv9yw, 2016-11-14:
I have alredy instaled PIO and integrated that in Atom. Could it be the cause of issue ?

gepd, 2016-11-14:
I think it's not the problem could be the language of your machine 

`""D:\arduino\soft\Sublime\Data\Packages\User\Deviot\penv\Scripts\pio"" ­Ґ пў«пҐвбп ў­гваҐ­­Ґ© Ё«Ё ў­Ґи­Ґ©
Є®¬ ­¤®©, ЁбЇ®«­пҐ¬®© Їа®Ја ¬¬®© Ё«Ё Ї ЄҐв­л¬ д ©«®¬.`  

this line seems to be weird, it needs a deeper analysis, if you want can contact me at gepd@outlook.com and we can schedule a teamviewer meeting. Is what I can do to try to resolve the issue because I can't replicate it

stanleyseow, 2017-02-03:
I have the same setup as m-chichikalov "Windows 7 64bit, Sublime Text 3126"
Same issue, I am on an US English version..

20:26:03 Extracting files...
20:26:03 Installing PlatformIO...
20:26:03 Error Creating the virtualenv

bkrajendra, 2017-03-31:
I'm also getting same error on Sublime text 3, Windows 10 64 bit.
Is there any solution.?
```
[ Deviot 1.2.5 ] Setup
14:21:23 PlatformIO is not installed
Preparing to installing it, please be patient it may take a while.
14:21:23 Extracting files...
14:21:23 Installing PlatformIO...
14:21:30 Error Creating the virtualenv

```

gepd, 2017-07-04:
can you check if this issue keeps present in deviot 2?

m-chichikalov, 2017-07-04:
Sorry, I cannot. I updated windows to 8.1 and started using atom instead of.
Have a nice one.

closed 2017-07-04 by gepd