gepd/Deviot #88: deviot library not found; platformio command line works

jwhendy, 2016-11-21:
This may be similar to #18 . I started playing with the teensy audio library + leds, and am using the following in my sketch:

```
#include <FastLED.h>
#include <SD.h>
#include <SPI.h>
#include <Wire.h>
#include <Audio.h>
```

From `subl3` + `deviot`, I get errors for missing header libraries, but via `platformio` directly, it works fine. Here's the `deviot` output:
```
[ Deviot 1.2.5 ] read-aud.ino
15:50:47 Building the project | Processing...
/home/jwhendy/Arduino/teensy-aud/lib/Audio/analyze_fft1024.cpp:28:26: fatal error: sqrt_integer.h: No such file or directory
#include "sqrt_integer.h"
^
/home/jwhendy/Arduino/teensy-aud/lib/Audio/analyze_fft256.cpp:28:26: fatal error: sqrt_integer.h: No such file or directory
#include "sqrt_integer.h"
^
/home/jwhendy/Arduino/teensy-aud/lib/Audio/control_ak4558.cpp:9:18: fatal error: Wire.h: No such file or directory
#include "wire.h"
^
15:50:48 ERROR | it took 0.81s
```

Here's the platformio output:
```
$ platformio run --target upload
[Sun Nov 20 22:23:35 2016] Processing teensy31 (platform: teensy, board: teensy31, framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Converting read-aud.ino
Collected 86 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <SPI> v1.0
|-- <FastLED>
|   |-- <DmxSimple> v3.1
|   |-- <SoftwareSerial> v1.0
|-- <Wire> v1.0
|-- <Audio> v1.3
|   |-- <SerialFlash>
|   |   |-- <SPI> v1.0
|   |-- <SPI> v1.0
|   |-- <SD> v1.0.8
|   |   |-- <SPI> v1.0
|   |-- <Wire> v1.0
|-- <SD> v1.0.8
|   |-- <SPI> v1.0
Compiling .pioenvs/teensy31/src/read-aud.ino.o
In file included from /home/jwhendy/Arduino/teensy-aud/src/read-aud.ino:29:0:
/home/jwhendy/.platformio/packages/framework-arduinoteensy/libraries/FastLED/FastLED.h:17:21: note: #pragma message: FastLED version 3.001.001
#    pragma message "FastLED version 3.001.001"
^
Linking .pioenvs/teensy31/firmware.elf
Checking program size .pioenvs/teensy31/firmware.elf
text	   data	    bss	    dec	    hex	filename
86832	    172	  12500	  99504	  184b0	.pioenvs/teensy31/firmware.elf
Building .pioenvs/teensy31/firmware.hex
Uploading .pioenvs/teensy31/firmware.hex
Teensy Loader, Command Line, Version 2.1
Read ".pioenvs/teensy31/firmware.hex": 87004 bytes, 33.2% usage
Soft reboot performed
Waiting for Teensy device...
(hint: press the reset button)
Found HalfKay Bootloader
Read ".pioenvs/teensy31/firmware.hex": 87004 bytes, 33.2% usage
Programming.....................................................................................
Booting
```

For my directory structure, here's what I have:

```
$ ls -lR
.:
total 20
drwxr-xr-x 2 jwhendy jwhendy 4096 Nov 20 15:50 lib
drwxr-xr-x 2 jwhendy jwhendy 4096 Nov 20 22:23 src
-rw-r--r-- 1 jwhendy jwhendy  411 Nov 20 14:38 platformio.ini
-rw-r--r-- 1 jwhendy jwhendy 4536 Nov 20 22:23 read-aud.ino

./lib:
total 0
lrwxrwxrwx 1 jwhendy jwhendy 37 Nov 20 15:50 Audio -> /home/jwhendy/Arduino/libraries/Audio

./src:
total 0
lrwxrwxrwx 1 jwhendy jwhendy 15 Nov 20 14:48 read-aud.ino -> ../read-aud.ino
```

From previous work with `deviot`, it seems it can find the library somehow (looking in the main `~/Arduino/libraries` folder?), but `platformio` wants them in `lib`, so I just symlink to what I want to bring in (hence, `Audio` just points to the version in the main Arduino folder). I have my main `.ino` file in the top-level directory but symlink to it in `src` to make `platformio` happy from the command line.

Let me know if you have any insights. This is on arch linux.

gepd, 2016-11-23:
Since Platformio 3 is possible to add an extra folder to find other libraries. 

The develop branch has that option already implemented, follow this steps to manually install it:

- Download [this](https://github.com/gepd/Deviot/archive/develop.zip) files (from the develop branch) and replace with the file in the `Packages/Deviot` 
- Remove `Main.sublime-menu` in the same folder to get the new changes in the menu
- Restart ST

Now you can select an extra folder in `Deviot Menu > Library Options > Add Extra Library Folder`

Let me know if that solves your problem

jwhendy, 2016-11-23:
I'll give this a try but am still a little confused. Is Deviot purely (mainly?) a sublime front end to platformio? I ask since platformio doesn't require me to specify another library folder, or at least symlinking to a needed library in lib/ seems to work.

What are the mechanics of what I'm doing by telling Deviot about an additional library folder? Is this applying a change to platformio behind the scenes or to Deviot?

Sorry for being confused. I'm not super familiar with the inner workings of any of this so it helps to have a gist of what's going on through these steps.

Thanks!

gepd, 2016-11-23:
Yes, Deviot is mainly a fron't end plugin wich handle PlatformIO, there some cases where Deviot make some thing automatically and in platformio need to do manually

for example, the feature to add a new folder where you have all your libraries need to be done manually in the `platformio.ini` file (http://docs.platformio.org/en/stable/projectconf.html#lib-dir), but Deviot do automatically in each file you are working on.

I think I missed something from your first message, now I understand, if you compile you code directly from platformio it do it without problem but from deviot show missing library. The answer is, it shouldn't shown that erros.

Are you using the same folder structure to compile in Deviot? some days ago someone tell me about the same error but it was because he has more than one file in the same folder, normally platformio compile all inside the `src` folder. Deviot can compile that way or using the `src_dir ` option wich override the `src` folder.

Can you add the structure of your folders to try to replicate the issue (including the libraries)

jwhendy, 2016-11-23:
Gotcha, and correct -- it works in `platformio`, but not in `Deviot`. Do you need more than what I posted above? I ended with the output of `ls -lR`, which shows that I have `src` containing a symlink to the top level `.ino` file (in case `Deviot` needed to look in the top-level), and a symlink to the teensy Audio library inside `lib`. That's it. 

Give me a bit and I'll create a reproducible example, though all I'm going to do is make a new dir, a `.ino` file with includes and empty `setup()` and `loop()`, and git pull a fresh `Audio` teensy copy. This is the first library that's complained and the behavior is not clear to me. Some libraries work as-is, others (like `FastLED` require symlinking into `lib`, so I'm not sure where `platformio` is looking for things...

jwhendy, 2016-11-25:
Further development... I looked at the output of `ps ax | grep subl` when `deviot` was hung trying to upload and see this:

```
/home/jwhendy/.config/sublime-text-3/Packages/User/Deviot/penv/bin/python /home/jwhendy/.config/sublime-text-3/Packages/User/Deviot/penv/bin/platformio -f -c sublimetext run -t upload -e teensy31
```

I wondered if the `platformio` bin used was different than my own, so I tried using yours directly vs. what I have from the Arch `platformio` package (`/usr/bin/platformio`). This works fine:

```
/home/jwhendy/.config/sublime-text-3/Packages/User/Deviot/penv/bin/platformio run -t upload
```

I can run that whole line (including the call to `.config/.../python` before it) as well.

Why would that be? What `subl3` appears to be calling in the background while trying to upload works directly, but not from within `subl3`? Hope that adds more information that can help track things down.

gepd, 2017-06-26:
This is fixed in Deviot 2. 

If you keep with this problem reopen this issue

closed 2017-06-26 by gepd

jwhendy, 2017-08-27:
I can't seem to open this, but it's probably not necessary. This is just to help anyone else who comes across confusion about this. I think there were a couple things going on:

- I *thought* I had instances where platformio could find included libraries without me doing anything. It's been a while, so I'm not trusting my memory. That said, there is [support that this works](https://github.com/platformio/platformio-core/issues/195)
- regardless of which case was going on above, I definitely didn't understand how platformio worked with respect to libraries. My solution was to `ln -s ~/Arduino/libraries/desired-lib ./lib/desired-lib` from inside my sketch folder
- from reading [the docs](http://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-extra-dirs), I now believe this should be done via `platformio.ini` and adding the `lib_extra_dirs` option.
- now that I realized *that*, I finally understand the comment above about `Deviot -> Library options -> Add extra library folder`!

So... we're all good for the most part. I think my comments in #154 stand though. It seems that you're re-creating what `platformio` already does, but in `/tmp`. I'm a hobby programmer, so I often just copy and paste my own code/files when I start on a new project as I re-use the includes. With `platformio` directly, I can do his no problem as my options from `platformio.ini` are carried over. With Deviot, it appears I have to reset the board, environment, and extra libraries because they stay in `/tmp` vs. with my project dir...

Anyway, thanks for your patience/persistence with me! I think there's room for improvement, but now very much believe this issue was my total mis-understanding!