gepd/Deviot #93: Unable to program Attiny through Arduino as ISP

hennenzac, 2016-12-08:
My problem is very similar to issue #40.  I am able to program the Attiny through the Arduino using the Arduino IDE.  I then programmed the Arduino alone using Deviot to see if I had that working, which was successful. So my setup should be correct.  I'm not sure what else to try.

 Here is the information in the console when I upload.

[consoleoutput.txt](https://github.com/gepd/Deviot/files/638662/consoleoutput.txt)

gepd, 2016-12-10:
In your output file, this line said the port cannot be find

`avrdude: ser_open(): can't open device "\\.\com1": The system cannot find the file specified.`

are you sure that is right port? 
Or may be the port may be in use by another program

hennenzac, 2016-12-11:
I saw that as well.  The Arduino is hooked up to COM4, which is what I chose in the Deviot menu/Serial ports.  At the top of the output file I sent it says Upload port: COM4.  Is there some other setting I need to change?  I have the board and environment set to Generic Attiny 85, Programmer set to Arduino as ISP (which i programmed to be an ISP via the Arduino IDE), and serial port set to COM4.

gepd, 2016-12-11:
Change the programmer to `None` and try again, it should solve the problem

hennenzac, 2016-12-11:
I changed it to None, but it still fails with a different error this time.  It says it can't find a USBtiny device.  I programmed the Arduino as ISP through the arduino IDE example, if that offers more info. 

Here is the last bit with the error.

Checking program size
text	   data	    bss	    dec	    hex	filename
1084	      0	      9	   1093	    445	.pioenvs\attiny85\firmware.elf
Building .pioenvs\attiny85\firmware.hex
Looking for upload port...
Uploading .pioenvs\attiny85\firmware.hex
avrdude: Error: Could not find USBtiny device (0x1781/0xc9f)

avrdude done.  Thank you.

*** [upload] Error 1
 [ERROR] Took 4.73 seconds 

 [SUMMARY] 
Environment uno     	[SKIP]
Environment attiny85	[ERROR]
 [ERROR] Took 4.73 seconds )
16:18:38 ERROR | it took 5.52s

gepd, 2016-12-13:
I've saw this answer about `Error: Could not find USBtiny device (0x1781/0xc9f)` https://stackoverflow.com/a/32944999/3769629 

Can you try pressing the reset button as JanuskaE recommend?

hennenzac, 2016-12-15:
I am assuming the reset on the Arduino board.  I have tried every combination of this with no change to the output file.

gepd, 2017-07-04:
can you check if this problem still keeps in deviot 2?

closed 2017-09-11 by gepd