gepd/Deviot #94: develop install failing

tablatronix, 2016-12-09:
installed develop, removed users/deviot
when I startup

[ Deviot 1.2.5 ] Setup
11:59:22 PlatformIO is not installed
Preparing to installing it, please be patient it may take a while.
11:59:22 Extracting files...
11:59:22 Installing PlatformIO...

console
```
Exception in thread Thread-4:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "...sublimetext\Data\Packages\Deviot\libs\Install.py", line 174, in threadcheckPio
    self.install()
  File "...sublimetext\Data\Packages\Deviot\libs\Install.py", line 272, in install
    if(out[0] > 0 or int(py_version) == 300):
ValueError: invalid literal for int() with base 10: ''
```

I have py 2.72

tablatronix, 2016-12-09:
I also tried this just to see what happens
Clicked install develop PIO, said installed
but this in console
```
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "C:\Users\alverson\Documents\projects\bin\sublimetext\Data\Packages\Deviot\libs\Install.py", line 109, in threadcheckPio
    Tools.getJSONBoards()
  File "C:\Users\alverson\Documents\projects\bin\sublimetext\Data\Packages\Deviot\libs\Tools.py", line 577, in getJSONBoards
    PlatformioCLI().getAPIBoards()
  File "C:\Users\alverson\Documents\projects\bin\sublimetext\Data\Packages\Deviot\libs\PlatformioCLI.py", line 720, in getAPIBoards
    boards = Run.runCommand(command, setReturn=True)
  File "C:\Users\alverson\Documents\projects\bin\sublimetext\Data\Packages\Deviot\libs\Commands.py", line 125, in runCommand
    self.resultsOutput(return_code)
  File "C:\Users\alverson\Documents\projects\bin\sublimetext\Data\Packages\Deviot\libs\Commands.py", line 297, in resultsOutput
    self.message_queue.put(message, current_time, diff_time)
AttributeError: 'CommandsPy' object has no attribute 'message_queue'

Package Control: Skipping automatic upgrade, last run at 2016-12-09 11:11:31, next run at 2016-12-09 12:11:31 or after
```

tablatronix, 2016-12-09:
ok so i just commented out that block

```
        if(out[0] > 0 or int(py_version) == 300):

            current_time = time.strftime('%H:%M:%S')
            self.message_queue.put(
                "error_installing_env_{0}", current_time)
            return
```
and pio installs

so py_version seems to be empty...

also, how the hell do you debug plugins ? Is there not a simple console print available
global stdout ?

tablatronix, 2016-12-09:
What is the purpose of this ?

```
        cmd = ['python', 'virtualenv.py', '"%s"' % self.env_dir]
        out = Tools.runCommand(cmd, virtualenv)
        print (cmd)
```

Seems to not return anything in cmd[1]

tablatronix, 2016-12-09:
```
New python executable in ...sublimetext\Data\Packages\User\Deviot\penv\Scripts\python.EXE
Installing setuptools, pip, wheel...done.
```

tablatronix, 2016-12-09:
Ok so im new to python. It looks like this is checking for errorlevel OR maybe an error about need python 300.

But the string to int fails when the string is empty, not sure though if it is allowed to be empty or not.

gepd, 2016-12-10:
I'm sorry about the delay, I''ve been busy this days...

That line you posted is installing the virtualenv where PlatformIO will be installed in case isn't installed. The current master branch has a problem with that, are you using the master or the developer branch? 
in the last one it problem shouldn't be happening

tablatronix, 2016-12-10:
pretty sure it was develop, I used package add repo to install it.
and after bypassing, my menu has quick menu at top

gepd, 2016-12-10:
try this:

Change: 

`if(out[0] > 0 or int(py_version) == 300):` until return

for:

```python
try:
    if(out[0] > 0 or int(py_version) == 300):

        current_time = time.strftime('%H:%M:%S')
        self.message_queue.put(
            "error_installing_env_{0}", current_time)
        return
except:
    pass
```

tablatronix, 2016-12-10:
I will test when I get back to my windows machine on monday

tablatronix, 2016-12-12:
Yes that works.

closed 2017-06-27 by gepd