gepd/Deviot #95: develop osx examples not working

tablatronix, 2016-12-10:
```
Traceback (most recent call last):
  File "/Applications/Sublime Text.app/Contents/MacOS/sublime_plugin.py", line 797, in run_
    return self.run()
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot/DeviotStarter.py", line 296, in run
    self.MENU_LIST = Menu().createLibraryExamplesMenu()
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot/libs/Menu.py", line 228, in createLibraryExamplesMenu
    library_paths = Paths.getLibraryFolders(platform)
UnboundLocalError: local variable 'platform' referenced before assignment
```

tablatronix, 2016-12-10:
Threw a 
platform = "" at the beginning of
`createLibraryExamplesMenu`

not sure if its global or not, shrug

gepd, 2017-06-20:
This is already fixed, please read:  https://github.com/gepd/Deviot/issues/117

closed 2017-06-20 by gepd