gepd/Deviot #96: develop osx serial ports menu not working

tablatronix, 2016-12-10:
when I click serial ports, it seems to get stuck in a loop with zero conf and msdns calls

```
    ----------------------------------------
Command ""/Users/shawn/Library/Application Support/Sublime Text 3/Packages/User/Deviot/penv/bin/python" -u -c "import setuptools, tokenize;__file__='/private/var/folders/0_/8qr6jgcx3qsgjcrpsgz2c9w00000gn/T/pip-build-upLL6W/netifaces/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /var/folders/0_/8qr6jgcx3qsgjcrpsgz2c9w00000gn/T/pip-Imb_YS-record/install-record.txt --single-version-externally-managed --compile --install-headers "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/User/Deviot/penv/include/site/python2.7/netifaces"" failed with error code 1 in /private/var/folders/0_/8qr6jgcx3qsgjcrpsgz2c9w00000gn/T/pip-build-upLL6W/netifaces/

Traceback (most recent call last):
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot/libs/mDNS.py", line 3, in <module>
    from zeroconf import ServiceBrowser, Zeroconf
ImportError: No module named zeroconf
```

tablatronix, 2016-12-10:
dup #72

tablatronix, 2016-12-28:
Reopening since #72 was closed

fix was to reinstall (missing) osx xcode command line tools

`xcode-select --install`

JayCarr, 2017-01-11:
I don't know if this might help, but I ended up having to open up XCode after a recent update, simply to okay the xcode-select license change....  Kinda defeats the whole purpose, but it did make it work.

tablatronix, 2017-01-11:
yeah I checked that, but didn't have anything like that. of course my xcode beta was older I think.
Good to know though.

gepd, 2017-06-30:
Fixed in v2.0, zeroconf was integrated to Deviot

closed 2017-06-30 by gepd