gepd/Deviot #102: raise error linter failing?

tablatronix, 2017-01-20:
I am trying to build an example file from a library, and nothing happens.

trace is
```
Exception in thread Thread-9:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "C:\Users\user\Documents\projects\bin\sublimetext\Data\Packages\Deviot\libs\PlatformioCLI.py", line 118, in fileCheck
    self.projectRecognition()
  File "C:\Users\user\Documents\projects\bin\sublimetext\Data\Packages\Deviot\libs\PlatformioCLI.py", line 151, in projectRecognition
    self.beforeProcess()
  File "C:\Users\user\Documents\projects\bin\sublimetext\Data\Packages\Deviot\libs\PlatformioCLI.py", line 214, in beforeProcess
    INIFILE = ConfigObj(C['INIPATH'], list_values=False)
  File "C:\Users\user\Documents\projects\bin\sublimetext\Data\Packages\Deviot\libs\configobj\configobj.py", line 1198, in __init__
    self._load(infile, configspec)
  File "C:\Users\user\Documents\projects\bin\sublimetext\Data\Packages\Deviot\libs\configobj\configobj.py", line 1288, in _load
    raise error
  File "<string>", line None
Deviot.libs.configobj.configobj.ConfigObjError: Parsing failed with several errors.
First error at line 16.

```

deviot console never comes up

gepd, 2017-06-07:
what was the name of the library? I know that you may solve this issue but I'm working in the code again

tablatronix, 2017-06-07:
Not sure, ill see if it happens again

gepd, 2017-06-30:
You can re-open it if it happens again

closed 2017-06-30 by gepd