gepd/Deviot #103: device not found for OTA upload

macmeck, 2017-01-20:
I have an ESP8266 which is ready for OTA upload. But when I select "Serial Port(s)", all I see is my COM Port and "Add IP Manually". Providing the IP does not help - upload does not start.

Board is ESP01_1M
Deviot v1.2.5
Pio v3.2.1

Using the Arduino IDE, OTA works just fine and the device is being detected automatically.

pxds, 2017-01-21:
I have the same problem as you, "Add IP Manually" does absolutely nothing. The OTA port still shows up and works on Arduino IDE.

Board NodeMCU 1.0
Deviot 1.2.5
PIO 3.2.1
Mac OSX El Captain 10.11.3

macmeck, 2017-02-03:
Is there anything we can do to track down this issue? It does not seem to be a Wifi issue since the Arduino IDE is working just fine. 
Any config option that needs to be set? Any other idea?

pxds, 2017-02-03:
I use a WiFi signal extender at home, once i turned it off, NodeMCU started showing much more often as an OTA port, but sometimes it would still fail the upload or disappear from the port list. 

Still clueless on how to solve it.

macmeck, 2017-02-03:
Well, with current development branch, at least setting the IP address manually seems to be working. Still no automatic discovery of the OTA port though.

marcunzueta, 2017-02-15:
@macmeck I have the exact same problem. I tested my ESP8266 on Deviot, Sublime Text 3 and I cannot upload by OTA as it is not detected on the serial port, even if I manually introduce the IP; but if I use the Arduino IDE it detects automatically the ESP8266 board and I can upload it without anyproblems

gepd, 2017-06-25:
Hey guys there is a new version of Deviot with improvements in this subject, it has  not officially release yet, but you can install it following the next instructions: https://github.com/gepd/Deviot/issues/117

I'll thank any feedback

Reopen this topic if you think it's neccesary

closed 2017-06-25 by gepd

macmeck, 2017-08-12:
took a while for me to test, but seems to be fine now! 👍