gepd/Deviot #105: IndexError: list index out of range

fthylmz, 2017-02-04:
```
Exception in thread Thread-10:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "F:\Users\###\AppData\Roaming\Sublime Text 3\Packages\Deviot\libs\Install.py", line 141, in threadcheckPio
    out = Tools.runCommand(cmd)
  File "F:\Users\###\AppData\Roaming\Sublime Text 3\Packages\Deviot\libs\Tools.py", line 814, in runCommand
    output = process.communicate()
  File "./python3.3/subprocess.py", line 920, in communicate
  File "./python3.3/subprocess.py", line 1233, in _communicate
IndexError: list index out of range

```

I can't install Deviot on Sublime Text 3 and I could not find any solution to get rid of this error. Do I need to provide anything else on this error?

To see complete console log click [here](http://pastebin.com/3iUpJMA1).

gepd, 2017-06-08:
You may have desist with Deviot, but can you tell me wich language had your machine when you tried this?

gepd, 2017-06-20:
Please re open this issue if you need

closed 2017-06-20 by gepd