gepd/Deviot #106: Nothing happen when I click Select board

stanleyseow, 2017-02-09:
My environment is as below ... 
OS X 10.10.5
Sublime 3126
Python 2.7.10

halfnibble, 2017-02-20:
I'm have the same issue with the latest version of Mac OS, Python 3.5.2 is my default, but I have Python 2.7.

halfnibble, 2017-02-20:
I just set my path to make Python 2.7.11 default. And I'm also on SublimeText 3126.

Uninstall and reinstall. Nothing—no boards to select. Your "Stino" plugin works perfectly though.

Another interesting note: when Deviot is installed, SublimeText fails to check for updates. It is timing out. But when I remove Deviot, SublimeText can check for updates.

halfnibble, 2017-02-20:
Ok. I think this may just be a UX issue then.

Apparently, after I created a new sketch, I was able to select a board. I guess that's the problem with being the type of person to configure everything before starting....

halfnibble, 2017-02-20:
But Deviot still prevents SublimeText 3126 from being able to check for updates. So maybe I should create a new issue for that.

tablatronix, 2017-03-19:
check some of my issues, I have osx and have resolved some issues manually myself. They are still open issues with no PRs. I had several problems i had to fix in code to get it working, are you using stable or dev ?

gepd, 2017-06-08:
I've take note of the sublime update issue.

The "issue" selecting a board when none is sketch is open happens because platformio works with one file config for each sketch, so currently only select a board when the config file is available, but I think I can work with something to improve this behavior

gepd, 2017-06-18:
Implemented.

closed 2017-06-18 by gepd