gepd/Deviot #107: Command line multiple

Clemzo, 2017-02-15:
Hello,

With Arduino IDE the command line multiple works correctly :
snprintf ( temp, 400,
"<html>\
  <head>\
    <title>Visit server</title>\
    <style>\
      body { background-color: #cccccc; font-family: Arial, Helvetica, Sans-Serif; Color: #000088; }\
    </style>\
  </head>\
  <body>\
    <h1>Hello from ESP8266!</h1>\
  </body>\
</html>"
  );
But Deviot return an error for each line:
warning: missing terminating " character [enabled by default]

Is It possible to have same comportment as Arduino IDE ?

Thanks

gepd, 2017-06-20:
> warning: missing terminating " character [enabled by default]

That is only a warning it's not stopping the compiling, please post the full error

closed 2017-06-20 by gepd

Clemzo, 2017-06-21:
Here is the full error :
`[ Deviot 1.2.5 ] test-visit_server-3.ino
08:15:59 Chargement du firmware | D:/Mes documents/Arduino/Mes scripts/ESP8266/test_visite_server/test-visit_server-3/test-visit_server-3.ino:60:1: warning: missing terminating " character [enabled by default]
"<html>\
^
D:/Mes documents/Arduino/Mes scripts/ESP8266/test_visite_server/test-visit_server-3/test-visit_server-3.ino:70:8: warning: missing terminating " character [enabled by default]
</html>"
^
D:/Mes documents/Arduino/Mes scripts/ESP8266/test_visite_server/test-visit_server-3/test-visit_server-3.ino:60:1: warning: missing terminating " character [enabled by default]
"<html>\
^
D:/Mes documents/Arduino/Mes scripts/ESP8266/test_visite_server/test-visit_server-3/test-visit_server-3.ino:60:1: error: missing terminating " character
D:/Mes documents/Arduino/Mes scripts/ESP8266/test_visite_server/test-visit_server-3/test-visit_server-3.ino:61:8: warning: missing terminating " character [enabled by default]
<head>\
^
D:/Mes documents/Arduino/Mes scripts/ESP8266/test_visite_server/test-visit_server-3/test-visit_server-3.ino:61:1: error: missing terminating " character
<head>\
^
D:/Mes documents/Arduino/Mes scripts/ESP8266/test_visite_server/test-visit_server-3/test-visit_server-3.ino: In function 'void handleRoot()':
D:/Mes documents/Arduino/Mes scripts/ESP8266/test_visite_server/test-visit_server-3/test-visit_server-3.ino:61:1: error: expected primary-expression before '<' token
D:/Mes documents/Arduino/Mes scripts/ESP8266/test_visite_server/test-visit_server-3/test-visit_server-3.ino:61:2: error: expected primary-expression before '/' token
<head>\
^
D:/Mes documents/Arduino/Mes scripts/ESP8266/test_visite_server/test-visit_server-3/test-visit_server-3.ino:61:3: error: 'html' was not declared in this scope
<head>\
^
D:/Mes documents/Arduino/Mes scripts/ESP8266/test_visite_server/test-visit_server-3/test-visit_server-3.ino:62:3: error: expected primary-expression before ')' token
<title>visit server</title>\
^
08:16:03 ERREUR | Temps : 3.64s`

And compiling is stopped