gepd/Deviot #108: deviot doesn't see homebrew's platformio?

thekelvinliu, 2017-02-16:
when i try to install deviot via package control, the window showing the install status says that platformio is not found.
it then starts to download and install platformio by itself.
however, i do have platformio installed via homebrew;
it's installed at `/usr/local/bin/platformio`.
i can't seem to find any place where i can manually set the path for this.
is this expected behavior?

gepd, 2017-06-20:
I've improved this feature in the in the last version of the plugin, it's not released yet but you can help me to test it, more information here: https://github.com/gepd/Deviot/issues/117

gepd, 2017-06-20:
Please re open this issue if you need

closed 2017-06-20 by gepd