gepd/Deviot #109: Deviot not detecting PlatformIO

Kenny11CZ, 2017-02-27:
Hi there, I've had problem with installing Deviot because of this error:
`ERROR: The executable "D:\Programs\Sublime Text 3\Data\Packages\User\Deviot\penv\Scripts\python.exe" could not be run: [Error 5] Access is denied`
which appeared in PlatformIO installation, so I tried install PlatformIO by myself, but Deviot do not detect that installation and it it fails again during installation.

Both errors were mentioned here, but no solution worked for me.

I would say the problem is this line in console log:
`'"D:\Programs\Sublime Text 3\Data\Packages\User\Deviot\penv\Scripts\pio"' is not recognized as an internal or external command,operable program or batch file.`

Additional informations:
No Atom installed, PlatformIO core only.
OS: Windows 10
ST Build: 3124
Non-portable version

**platformio --version**
`PlatformIO, version 3.2.1`

**pio --version**
`PlatformIO, version 3.2.1`

**python --version**
`Python 2.7.12`

**virtualenv --version**
`15.0.3`

**full console log**
`DPI scale: 1.25
startup, version: 3126 windows x32 channel: stable
executable: /D/Programs/Sublime Text 3/sublime_text.exe
working dir: /D/Programs/Sublime Text 3
packages path: /D/Programs/Sublime Text 3/Data/Packages
state path: /D/Programs/Sublime Text 3/Data/Local
zip path: /D/Programs/Sublime Text 3/Packages
zip path: /D/Programs/Sublime Text 3/Data/Installed Packages
ignored_packages: ["Vintage"]
pre session restore time: 0.617538
startup time: 0.681538
first paint time: 0.699538
first paint time: 0.699538
no such context .regex..dart in Packages/Dart/Dart.tmLanguage
reloading plugin Default.auto_indent_tag
reloading plugin Default.block
reloading plugin Default.comment
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.install_package_control
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.quick_panel
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
[... 262 lines cut ...]
WINDOW COMMAND ENABLED True
WINDOW COMMAND ENABLED True
WINDOW COMMAND ENABLED True
WINDOW COMMAND ENABLED True
WINDOW COMMAND ENABLED True
WINDOW COMMAND ENABLED False
ignored packages updated to: ["Deviot", "Vintage"]
reloading settings Packages/User/Package Control.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings
reloading settings Packages/User/Package Control.sublime-settings
ignored packages updated to: ["Vintage"]
reloading plugin Deviot.DeviotStarter
generating syntax summary
no such context .regex..dart in Packages/Dart/Dart.tmLanguage
generating meta info summary
'"D:\Programs\Sublime Text 3\Data\Packages\User\Deviot\penv\Scripts\pio"' is not recognized as an internal or external command,
operable program or batch file.

reloading settings Packages/User/Package Control.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings
indexing: crawldb not available, indexing abandoned
New python executable in D:\Programs\Sublime Text 3\Data\Packages\User\Deviot\penv\Scripts\python.exe
ERROR: The executable "D:\Programs\Sublime Text 3\Data\Packages\User\Deviot\penv\Scripts\python.exe" could not be run: [Error 5] Access is denied
`

gepd, 2017-06-20:
Hi @Kenny11CZ , I've improved this feature in the in the last version of the plugin, it's not released yet but you can help me to test it, more information here: https://github.com/gepd/Deviot/issues/117

Kenny11CZ, 2017-06-20:
Nice, Deviot is now successfuly installed, but I don't have any project at this moment to test it on...

gepd, 2017-06-20:
That is good to know, you can report an issue any time!

closed 2017-06-20 by gepd