gepd/Deviot #113: How to access the virtualenv to run platformio commands?

garvamel, 2017-05-11:
Not from Sublime/Deviot, just from terminal. Is it possible? I don't have experience working with virtualenvs.

Thanks!

airdrummingfool, 2017-05-20:
I was able to install platformio via Homebrew (I'm on a Mac). Using that version, I was able to install new platforms, which I was then able to use in Deviot. Perhaps you can do something similar and not have to worry about accessing the platformio in the virtualenv.

gepd, 2017-06-07:
Is not possible to this from a easy way, I recommend you to install platformio from your console. 

Using pip you need to run this command in your terminal 

`pip install -U platformio`

When it finished, you can run the platformIO commands from there

gepd, 2017-06-20:
@garvamel I have released a beta of Deviot 2 where you can access to the platformio command from the terminal (Menu > Open Terminal)

Follow this steps to install it: https://github.com/gepd/Deviot/issues/117

closed 2017-06-20 by gepd