gepd/Deviot #114: Console error links broken

thinkyhead, 2017-05-18:
I'm using the current stable DevIoT on macOS with the current Sublime Text 3. My issue pertains to the repository https://github.com/MarlinFirmware/Marlin if you want to see what I'm talking about…

The `platform.ini` file is installed at the top level, while the `.ino` and other source files are in a subfolder called `Marlin`. This problem only occurs only when I have `platform.ini` file at the same level as the folder designated by `src_dir` (`Marlin`). It doesn't occur if `platform.ini` is inside the `Marlin` folder at the same level as the `Marlin.ino` file with `src_dir = .`.

What happens is that after a build, links to errors in the output console have the wrong path. For example, on error it will "open" `./Marlin/Marlin/SanityCheck.h` and `./Marlin/Marlin/Marlin/SanityCheck.h`, while the link in the console points to `./Marlin/Marlin/Marlin/SanityCheck.h`. I've tried playing with the `.platform.ini` settings and DevIoT "Default Path", but no luck.

I'll take a look at the DevIoT code and see if anything obvious is going. If I can figure out a fix I will submit it.

thinkyhead, 2017-05-19:
I've made some progress in chasing this down. Evidently when `platformio.ini` is in the `Marlin` folder with `src_dir = .` the console output includes the full paths to the files that are being flagged. However, when `platformio.ini` is in the folder above `Marlin` with `src_dir = Marlin` (or `./Marlin`) console output only includes the relative path, as in `Marlin/filename.cpp`.

So in this case, when you click on a line containing a filename and line numbers, DevIoT uses the path to the currently active file in the front window. And, it responds most often twice to the `on_selection_modified` event. This is why two phantom files are opened: one with `Marlin/Marlin/filename.cpp` and the next with `Marlin/Marlin/Marlin/filename.cpp`.

Interestingly, if the active file is in the `Marlin` folder, `Tools.getWorkingPath` returns the path to the folder above. But for files in subfolders of `Marlin`, `Tools.getWorkingPath` returns the path to the folder of the active file, which includes `Marlin` as one of the path elements.

Anyway, all I really need is some reliable way to get the path to the active `platformio.ini` file and/or the `src_dir` value from `platform.ini` so I can use it in `on_selection_modified` to synthesize the correct path. If there's already an accessor function for that data, I would prefer to use it. Otherwise, I'm liable to write a function that starts from the `workingPath` and goes up the hierarchy looking for the file.

gepd, 2017-06-18:
I'm sorry for the long delay in my answer @thinkyhead I've been a little busy working in the version 2.0 of the plugin. I've started to write it from the scratch, now it's more organized.

There are still some features I'm working on, but the main ones (compile, upload, etc) are ready.

I didn't understand very well you issue, but first, let me explain some things; PlatformIO has is own way to organize the files, all the files always must be in the `src` folder and the `platformio.ini` always in the parent folder. That is the way the program originally works. I've started to use the `src_dir` option to override the folder where you put your sketch file(s), that is if you don't like all the extra files generated by platformio.

As the code of the current version is so messy, I'll avoid to tell you how to get the `platformio.ini` file but I hope to release in the nexts hours a dev version. In that version is much more easy to get information like platformio.ini path, working directory, file name, etc. I'm currently working in the plugin, so I can release a fast fix in case to need it.

So if you can try to explain me again what is the issue and we can work on it together.

thinkyhead, 2017-06-19:
Marlin adopted PlatformIO as an additional build option while retaining its existing file layout, because PlatformIO doesn't _require_ the `src` folder layout. If you download [the current Marlin](https://github.com/MarlinFirmware/Marlin/archive/bugfix-1.1.x.zip) and build it with Deviot you will see that it builds successfully even though it doesn't use the `src` layout.

If you insert intentional errors into the code (e.g., `#error "Hey"` in `Marlin_main.cpp`) you will see that the console displays relative paths in the error messages. As described above, this behavior depends on the value of `src_dir`.

Anyway, there is good news. The patch I recently submitted (#115) includes the fix for this issue. If you're using the latest Deviot including #115, then clicking on an error should now do the correct thing.

closed 2017-06-19 by thinkyhead

gepd, 2017-06-20:
I'll figure out your fix to implement it in the new code because, the current one, will be outdated in the next release.

Thanks for your for your contribution!