gepd/Deviot #118: Command line multiple (re open closed issue #107)

Clemzo, 2017-06-21:
This issue stop the compiling because compiler is unable to interprete the rest of instructions.

Here is the full error :

```
[ Deviot 1.2.5 ] test-visit_server-3.ino
08:15:59 Chargement du firmware | D:/Mes documents/Arduino/Mes scripts/ESP8266/test_visite_server/test-visit_server-3/test-visit_server-3.ino:60:1: warning: missing terminating " character [enabled by default]
"<html>\
^
D:/Mes documents/Arduino/Mes scripts/ESP8266/test_visite_server/test-visit_server-3/test-visit_server-3.ino:70:8: warning: missing terminating " character [enabled by default]
</html>"
^
D:/Mes documents/Arduino/Mes scripts/ESP8266/test_visite_server/test-visit_server-3/test-visit_server-3.ino:60:1: warning: missing terminating " character [enabled by default]
"<html>\
^
D:/Mes documents/Arduino/Mes scripts/ESP8266/test_visite_server/test-visit_server-3/test-visit_server-3.ino:60:1: error: missing terminating " character
D:/Mes documents/Arduino/Mes scripts/ESP8266/test_visite_server/test-visit_server-3/test-visit_server-3.ino:61:8: warning: missing terminating " character [enabled by default]
<head>\
^
D:/Mes documents/Arduino/Mes scripts/ESP8266/test_visite_server/test-visit_server-3/test-visit_server-3.ino:61:1: error: missing terminating " character
<head>\
^
D:/Mes documents/Arduino/Mes scripts/ESP8266/test_visite_server/test-visit_server-3/test-visit_server-3.ino: In function 'void handleRoot()':
D:/Mes documents/Arduino/Mes scripts/ESP8266/test_visite_server/test-visit_server-3/test-visit_server-3.ino:61:1: error: expected primary-expression before '<' token
D:/Mes documents/Arduino/Mes scripts/ESP8266/test_visite_server/test-visit_server-3/test-visit_server-3.ino:61:2: error: expected primary-expression before '/' token
<head>\
^
D:/Mes documents/Arduino/Mes scripts/ESP8266/test_visite_server/test-visit_server-3/test-visit_server-3.ino:61:3: error: 'html' was not declared in this scope
<head>\
^
D:/Mes documents/Arduino/Mes scripts/ESP8266/test_visite_server/test-visit_server-3/test-visit_server-3.ino:62:3: error: expected primary-expression before ')' token
<title>visit server</title>\
^
08:16:03 ERREUR | Temps : 3.64s
```
Thank's

gepd, 2017-06-21:
Is it possible to get the full code to test it in my machine?

Clemzo, 2017-06-21:
Ok, here is the code : 

```
/*  test multilines
 *   Work on Nodemcu
 */

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
//#include <DNSServer.h>

const char *ssid = "maisonclem";
const char *password = "1234567890";
ESP8266WebServer server ( 80 );

void setup () {
	Serial.begin ( 115200 );
  Serial.println ("\n\n  test multilines");
  connexionWifi ();
}

void loop () {
	server.handleClient();
}

void handleRoot() {
  char temp[400];
  int sec = millis() / 1000;
  int min = sec / 60;
  int hr = min / 60;
  snprintf ( temp, 400,
"<html>\
  <head>\
    <title>Visit server</title>\
    <style>\
      body { background-color: #cccccc; font-family: Arial, Helvetica, Sans-Serif; Color: #000088; }\
    </style>\
  </head>\
  <body>\
    <h1>Hello from ESP8266!</h1>\
  </body>\
</html>"
  );
  server.send ( 200, "text/html", temp );
}

void handleNotFound() {
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += ( server.method() == HTTP_GET ) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for ( uint8_t i = 0; i < server.args(); i++ ) {
    message += " " + server.argName ( i ) + ": " + server.arg ( i ) + "\n";
  }
  server.send ( 404, "text/plain", message );
}

void connexionWifi () {
  WiFi.begin ( ssid, password );
  while ( WiFi.status() != WL_CONNECTED ) {                           // Wait for connection
    delay ( 500 );
    Serial.print ( "." );
  }
  Serial.printf ( "\nConnected to %s\n", ssid );
  Serial.printf ( "IP address: %s\n", WiFi.localIP().toString().c_str());
  server.on ( "/", handleRoot );
  server.onNotFound ( handleNotFound );
  server.begin();
  Serial.println ( "HTTP server started" );
}
```
Thank's

gepd, 2017-06-22:
```c++
/*  test multilines
 *   Work on Nodemcu
 */

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
//#include <DNSServer.h>

const char *ssid = "test";
const char *password = "123";
ESP8266WebServer server ( 80 );

void setup () {
    Serial.begin ( 115200 );
  Serial.println ("\n\n  test multilines");
  connexionWifi ();
}

void loop () {
    server.handleClient();
}

void handleRoot() {
  char temp[400];
  int sec = millis() / 1000;
  int min = sec / 60;
  int hr = min / 60;
  snprintf ( temp, 400,
  "<html>" \
    "<head>" \
      "<title>Visit server</title>" \
      "<style>" \
        "body { background-color: #cccccc; font-family: Arial, Helvetica, Sans-Serif; Color: #000088; }" \
      "</style>" \
    "</head>" \
    "<body>" \
      "<h1>Hello from ESP8266!</h1>" \
    "</body>" \
  "</html>"
    );
  server.send ( 200, "text/html", temp );
}

void handleNotFound() {
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += ( server.method() == HTTP_GET ) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for ( uint8_t i = 0; i < server.args(); i++ ) {
    message += " " + server.argName ( i ) + ": " + server.arg ( i ) + "\n";
  }
  server.send ( 404, "text/plain", message );
}

void connexionWifi () {
  WiFi.begin ( ssid, password );
  while ( WiFi.status() != WL_CONNECTED ) {                           // Wait for connection
    delay ( 500 );
    Serial.print ( "." );
  }
  Serial.printf ( "\nConnected to %s\n", ssid );
  Serial.printf ( "IP address: %s\n", WiFi.localIP().toString().c_str());
  server.on ( "/", handleRoot );
  server.onNotFound ( handleNotFound );
  server.begin();
  Serial.println ( "HTTP server started" );
}
```

That way is compiling to me

Clemzo, 2017-06-22:
Ok thanks for test but I don't where is the issue. The same sketch compile fine on Arduino IDE and found an error with Deviot (in my environnement).
Is there an issue in setup (ST3, Deviot,...) ?

gepd, 2017-06-22:
The issue is the concatenation of the html.

PlatformIO (who is the core of Deviot) is more strict than Arduino IDE to process the code, in this case, it works with double slash `\\` and the way as is now:

```
"Text" \
"More text"
```

Clemzo, 2017-06-22:
Ok I've put double slash at end of multilines and now I have lot of warning but not stop on ST and Arduino IDE, but that work well.
I'm sorry to work you on this.
You can closed this issue

gepd, 2017-06-22:
Yes, I've seen the warnings to, with the second option there are less warning.

Don't worry, every report help me to improve the plugin.

closed 2017-06-22 by gepd