gepd/Deviot #122: Is it normal that it takes 30 seconds to upload a sketch?

nicmar, 2017-06-29:
It takes 30-40 seconds to upload a sketch before it's being run. Is that the expected behaviour or is there anything I can do to make it faster?

It's kind of annoying to wait all the time.. :)

gepd, 2017-06-29:
It's never taken that long, except when you're using the platform for the first time and the tool chain must be downloaded and installed.

Can you tell me your OS, board used?
Are you using the master branch?

nicmar, 2017-06-30:
I'm using MacOS Sierra, with a NodeMCU ESP-12e. Maybe this takes longer time than on an Arduino? I also asked on the esp8266 forum (http://www.esp8266.com/viewtopic.php?f=160&t=15338) and they say it's normal. They also say that I should increase baud rate to 921600. From what I can see in Deviot, the max I can choose is 230400.

Is there a code verification going on after upload that can be turned off or somethine else?

gepd, 2017-06-30:
Yes, for ESP it's common, at least I have always got that amount of time, it seems to be faster when I work with OTA in my local network.

The only baudrate option at this moment in Deviot is for the serial monitor, but I'm interested in this topic so I'll check how fast it works with a higher baudrate and I'll add an option to override it when you upload an sketch.

I'm very close to release Deviot 2 if you want, you can install it (read this:) 
I'll let you know when it is added

gepd, 2017-06-30:
It's done, currently available in the V2 branch :)

closed 2017-06-30 by gepd