gepd/Deviot #123: feature serial console improvments

tablatronix, 2017-07-02:
Not sure if i mentioned this in another serial console improvements issue or not.

- [ ]  ability to set newline send characters crlf cr lf or simiar to putty implicit cr in lf lf in cr etc.
- [ ]  The ability to send control characters would also be nice. Can use putty shortcuts maybe. ctrl+l etc.

I often need to explicitly send a carriage return or newline or other control characters.

gepd, 2017-07-02:
What would be the difference with the options in `Monitor Serial > Line endinds`?

tablatronix, 2017-07-03:
Isnt that rx only

gepd, 2017-07-03:
it's tx only, look: https://github.com/gepd/Deviot/blob/master/libraries/serial.py#L162-L168

tablatronix, 2017-07-03:
hmm, ill test again in had a problem communicating with a device and i could not send proper newlines to it no matter what I tried.

gepd, 2017-07-03:
This is currently sending: https://github.com/gepd/Deviot/blob/master/presets/main_menu.json#L291-L309

gepd, 2017-09-12:
did you check this?

tablatronix, 2017-09-12:
I will be working on that project again soon, so I will check

gepd, 2019-02-17:
Any news on this?
At least on the first point

What do you think of a command to send any character?
it can include the end of line characters by default or always show the last character sent

tablatronix, 2019-02-17:
Like a special syntax or a seperate cmd sendline?
I thought an escape code etc /0x2c

gepd, 2019-02-18:
No I was wrong, it was the same as the send command.
I think shortcuts can be possible, what command will be essential to you?

I would like to include the most basic, but would be possible to add new ones pretty easy

tablatronix, 2019-02-19:
I will take a look at putty again it seemed to have the usuals figured out