gepd/Deviot #124: Can't install platformIO

C0br4, 2017-07-03:
Before report an issue, please make sure it hasn't been reported before.

### Operating system
macOS 10.12.5 with current sublime text 3 (Build 3126)

### Description of problem
Can't install / update PlatformIO via Deviot thus unable to compile / upload

### Steps to Reproduce
fresh sublime installation; installing Deviot via Package Control

Menu->Deviot->Options->Upgrade PlatformIO

### Actual Results

During installation of Deviot via Package Control console outputs error that platformIO could not be installed

when called via Menu ('Upgrade PlatformIO') "Searching for PlatformIO Updates..." appears in console but nothing happens afterwards

### Expected Results
a recent and complete installation of platformIO

### Additional info

*You can edit the text above according to your need*

gepd, 2017-07-03:
Please paste here the output of the Sublime Text console (`View > Show Console`) after you run the `Upgrade PlatformIO` option

C0br4, 2017-07-03:
Sorry, I'm not familiar with sublime but I think nothing related to the issue is being logged into the console

## Sublime Console output:
ignored packages updated to: ["Deviot", "Vintage"]
unloading plugin Deviot.Deviot
generating syntax summary
generating meta info summary
reloading settings Packages/User/Package Control.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings
Package Control: The package Deviot has been removed
reloading settings Packages/User/Package Control.sublime-settings
ignored packages updated to: ["Vintage"]
reloading settings Packages/User/Package Control.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings
ignored packages updated to: ["Deviot", "Vintage"]
reloading settings Packages/User/Package Control.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings
ignored packages updated to: ["Vintage"]
reloading settings Packages/User/Package Control.sublime-settings
reloading plugin Deviot.Deviot
generating syntax summary
generating meta info summary
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/Package Control.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings

## Deviot console output:
[ Deviot 2.0.0 ] Starting...
12:17:10 Searching for PlatformIO Updates...

gepd, 2017-07-03:
Did Deviot install PlatformIO at the beginning? or did you had PlatformIO already on your machine?

As I don't have access right now to a machine with macOS, I'll have to wait a few hours until get one and test this.

I'll back soon

C0br4, 2017-07-03:
I uninstalled Deviot because I thought something was 'wrong' (I know, not very specific, turned out it only updated itself to 2.0.0 where syntax highlighting isn't set right), so I had an installation from the former Deviot Version if it did not remove it during uninstall. But I never installed platformIO by hand.

C0br4, 2017-07-03:
I tried to install Deviot with a fresh macOs user account via PackageControl. It failed to install platformIO

## Deviot console output:
[ Deviot 2.0.0 ] Setup
12:45:31 PlatformIO is not installed
Preparing to install it. Please be patient it; may take a while.
12:45:31 Downloading necessary files...
12:45:42 Extracting files...
12:45:42 Installing PlatformIO...
12:45:48 error installing platformio

## Sublime console output:
reloading plugin Package Control.1_reloader
reloading plugin Package Control.2_bootstrap
reloading plugin Package Control.Package Control
Package Control: Not running package cleanup since bootstrapping is not yet complete
reloading settings Packages/Package Control/Package Control.sublime-settings
reloading plugin 0_package_control_loader.00-package_control
reloading settings Packages/User/Package Control.sublime-settings
ignored packages updated to: ["Deviot", "Vintage"]
reloading settings Packages/User/Package Control.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings
reloading settings Packages/User/Package Control.sublime-settings
ignored packages updated to: ["Vintage"]
reloading plugin Deviot.Deviot
generating syntax summary
generating meta info summary
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/Package Control.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings

schack-lindemann, 2017-07-03:
Having the same issues. Tried a fresh install of Sublime Dev v.3140 with same problem. After trying to upgrade PlatformIO Deviot hangs. Here's Deviot log and Console log:

[ Deviot 2.0.0 ] Setup
14:48:39 PlatformIO is not installed
Preparing to install it. Please be patient it; may take a while.
14:48:39 Downloading necessary files...
14:48:40 Extracting files...
14:48:40 Installing PlatformIO...
14:48:43 error installing platformio

startup, version: 3140 osx x64 channel: dev
executable: /Applications/Sublime Text.app/Contents/MacOS/Sublime Text
working dir: /
packages path: /Users/schack/Library/Application Support/Sublime Text 3/Packages
state path: /Users/schack/Library/Application Support/Sublime Text 3/Local
zip path: /Applications/Sublime Text.app/Contents/MacOS/Packages
zip path: /Users/schack/Library/Application Support/Sublime Text 3/Installed Packages
ignored_packages: ["Vintage"]
pre session restore time: 0.221879
using gpu buffer for window
startup time: 0.283671
reloading plugin Default.auto_indent_tag
reloading plugin Default.block
reloading plugin Default.comment
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.install_package_control
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.quick_panel
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.settings
reloading plugin Default.show_scope_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin Default.ui
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
plugins loaded
environment variables loaded using: /bin/bash -l
reloading plugin Package Control.1_reloader
reloading plugin Package Control.2_bootstrap
reloading plugin Package Control.Package Control
Package Control: Not running package cleanup since bootstrapping is not yet complete
reloading settings Packages/Package Control/Package Control.sublime-settings
reloading plugin 0_package_control_loader.00-package_control
reloading settings Packages/User/Package Control.sublime-settings
ignored packages updated to: ["Deviot", "Vintage"]
reloading settings Packages/User/Package Control.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings
reloading settings Packages/User/Package Control.sublime-settings
ignored packages updated to: ["Vintage"]
reloading plugin Deviot.Deviot
generating syntax summary
generating meta info summary
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/Package Control.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings

gepd, 2017-07-03:
Thanks for the report @schack-lindemann I'm working to fix this issue, I'll have news soon

gepd, 2017-07-03:
Please Install the develop branch (Following [this instructions](https://github.com/gepd/Deviot/wiki/Setup#developer))

And let me know if the last commit fixed your problem

nobios, 2017-07-03:
Works for me...

[ Deviot 2.0.1.dev ] Setup
15:07:24 PlatformIO is not installed
Preparing to install it. Please be patient it; may take a while.
15:07:24 Downloading necessary files...
15:07:24 Extracting files...
15:07:24 Installing PlatformIO...
15:07:31 Installation finished, now you can enjoy Deviot!

gepd, 2017-07-03:
Thanks! I'll release it now!

gepd, 2017-07-03:
I've release [v2.0.1](https://github.com/gepd/Deviot/releases/tag/v2.0.1)

Reopen this topic if you keep having problems

closed 2017-07-03 by gepd

schack-lindemann, 2017-07-03:
Works for me too! Thanks for an excellent tool :-)