gepd/Deviot #125: ino files syntax

C0br4, 2017-07-03:
### Operating system
 macOS 10.12.5 + Sublime Text Build 3126

### Description of problem
since 2.0.0 newly opened ino files are in plaintext syntax

### Steps to Reproduce
open ino file

### Expected Results
ino files should be loaded with C syntax

gepd, 2017-07-03:
Please select the syntax manually from the right down corner, I assume it will say `Text` at this moment, you should select `Deviot`. 

You should do this only one time, next time will be automatically

C0br4, 2017-07-03:
Strangely I don't have a Deviot entry in this list anymore. I'm sure it was there before Deviot 2.0.0

gepd, 2017-07-03:
Now I'm thinking the syntax file are created when PlatformIO setup finished, try `Options > Rebuild Syntax and Completions Files` after that you should be able to select from the list.

I'll improve this with the rest of the others bugs

C0br4, 2017-07-03:
Thanks, that fixed it.

Keep it up, I think Deviot has great potential. There are so many people tired of the terrible Arduino IDE.

closed 2017-07-03 by C0br4