gepd/Deviot #126: Compiling only works once

C0br4, 2017-07-04:
Before report an issue, please make sure it hasn't been reported before.

### Operating system
macOS 10.12.5 // Sublime Build 3126 // Deviot 2.0.0.1
### Board used
Arduino Pro Mini 5V 16MHz

### Description of problem
I'm only able to compile one time after starting Sublime. The second time I click on compile nothing happens (no deviot console output). After restarting Sublime I can compile exactly one time after the same problem occurs.

## sublime console output:
Traceback (most recent call last):
  File "/Applications/Sublime Text.app/Contents/MacOS/sublime_plugin.py", line 797, in run_
    return self.run()
  File "/Users/C0br4/Library/Application Support/Sublime Text 3/Packages/Deviot/commands/deviot_compile_sketch.py", line 6, in run
    Compile()
  File "/Users/C0br4/Library/Application Support/Sublime Text 3/Packages/Deviot/platformio/compile.py", line 20, in __init__
    super(Compile, self).__init__()
  File "/Users/C0br4/Library/Application Support/Sublime Text 3/Packages/Deviot/platformio/initialize.py", line 29, in __init__
    reset_phantoms()
  File "/Users/C0br4/Library/Application Support/Sublime Text 3/Packages/Deviot/libraries/tools.py", line 493, in reset_phantoms
    del_phantom(pname)
  File "/Users/C0br4/Library/Application Support/Sublime Text 3/Packages/Deviot/libraries/tools.py", line 477, in del_phantom
    view = VPHANTOMS[phantom_id]
KeyError: 'error39'
Traceback (most recent call last):
  File "/Applications/Sublime Text.app/Contents/MacOS/sublime_plugin.py", line 797, in run_
    return self.run()
  File "/Users/C0br4/Library/Application Support/Sublime Text 3/Packages/Deviot/commands/deviot_compile_sketch.py", line 6, in run
    Compile()
  File "/Users/C0br4/Library/Application Support/Sublime Text 3/Packages/Deviot/platformio/compile.py", line 20, in __init__
    super(Compile, self).__init__()
  File "/Users/C0br4/Library/Application Support/Sublime Text 3/Packages/Deviot/platformio/initialize.py", line 29, in __init__
    reset_phantoms()
  File "/Users/C0br4/Library/Application Support/Sublime Text 3/Packages/Deviot/libraries/tools.py", line 493, in reset_phantoms
    del_phantom(pname)
  File "/Users/C0br4/Library/Application Support/Sublime Text 3/Packages/Deviot/libraries/tools.py", line 477, in del_phantom
    view = VPHANTOMS[phantom_id]
KeyError: 'error39'

gepd, 2017-07-04:
Did the first time, your code shows some error? 
is it possible you can post your sketch?

C0br4, 2017-07-04:
Yes there were some errors in my Code.
"Unfortunately" everything works again as I fixed them. I'm going through my git history and try to reproduce the bug and post the sketch here.

gepd, 2017-07-04:
I'll appreciate that, it will be very useful

C0br4, 2017-07-04:
It seems that basically every error during compilation cause this. For example a not declared variable

```
#include <Arduino.h>

// const int note_A7 = 3520;
const int speakerPin = 9;

void setup(){
	pinMode(speakerPin, OUTPUT);
}

void loop(){
	beep(speakerPin, note_A7, 100);
}

void beep (int speakerPin, float noteFrequency, long noteDuration){
}

```

gepd, 2017-07-05:
Thanks, I finally got to reproduce the issue. It's solved, however you will note that some times the error window will be displayed in the wrong line. It's because PlatformIO can't handle very well `ino` files.

what is the main diferences in the `ino` files? that most of the users put the custom funtions after `setup` and `loop`, but the PlatformIO compiler is expecting the oposite.

I'll work in a new feature to detect that, warn the user and fix it automatically if they want, but will take some days. So If you want to avoid this behavior just put your `setup` and `loop` functions at the end.

Let me know if the last fix is working in your code.

C0br4, 2017-07-05:
Qith the develop version it seems to be fixed in my cases. Files that formerly only compiled once compile fine now (except they don't because they have errors :) ). If I encounter this again
I'll go with your proposed workaround for now, let me know if you need anyone testing new changes you might make.

C0br4, 2017-07-05:
I close it for now as it fixed my problem and I'm not able to reproduce this error anymore with the develop branch.

closed 2017-07-05 by C0br4