gepd/Deviot #127: Excessive CPU usage

m4rcu5, 2017-07-04:
Hi,

I reported this in #117, but as that one is currently closed, lets move it here.

When having the plugin installed, and no project loaded, the CPU seems to be excessively used. The culprit is in one of the `plugin_host` threads of `sublime_text`.
Tracing the thread, it shows lots of activity in the virtual env, lots of the Python modules are accessed, as well as the package configuration files.

This makes my CPU usage on my i5 laptop spikes 12-25%.

gepd, 2017-07-04:
I'll start to check this again... 
is this happening even if you open ST and let it there without doing nothing?

m4rcu5, 2017-07-05:
Hi @gepd,

I closed all files and restarted Sublime, it still uses the same amount of cpu. Letting it sit for a while does not change anything either.
Looking at the strace, it seems to do the same action every 2 seconds.

Thanks!

gepd, 2017-07-05:
mmm That so weird, I don't have any code in background like you mentioned. Yesterday I tested this with ProcessExplorer in Windows and nothing was in background, so with my configuration there is not problem.

I'll try to replicate your machine to see if can get the same results; Please tell me your O.S, ST version and plugins installed

m4rcu5, 2017-07-05:
Sorry, I should have added that before.

I have dumped the complete strace in a [gist](https://gist.github.com/m4rcu5/df3e1747e305cd0836ecc3ede1d4df8e).

The following plugins are installed (pulled from the profiler):
```
This list shows how much time each plugin has taken to respond to each event:

on_activated:
    SublimeCodeIntel.SublimeCodeIntel: 0.002s total, mean: 0.000s, max: 0.001s
    SublimeLinter.sublimelinter: 0.016s total, mean: 0.000s, max: 0.005s
    Default.pane: 0.001s total, mean: 0.000s, max: 0.000s
    GitGutter.modules.events: 0.004s total, mean: 0.000s, max: 0.001s
    SideBarEnhancements.SideBar: 0.002s total, mean: 0.000s, max: 0.001s
    Deviot.Deviot: 0.046s total, mean: 0.001s, max: 0.009s
    Git.git.ignore: 0.002s total, mean: 0.000s, max: 0.001s
    Git.git.statusbar: 0.024s total, mean: 0.001s, max: 0.005s
    Vintage.vintage: 0.009s total, mean: 0.000s, max: 0.005s

on_deactivated:
    Default.history_list: 0.034s total, mean: 0.001s, max: 0.005s
    Vintage.vintage: 0.011s total, mean: 0.000s, max: 0.003s

on_hover:
    Default.symbol: 0.004s total, mean: 0.001s, max: 0.004s
    GitGutter.modules.events: 0.000s total, mean: 0.000s, max: 0.000s

on_modified:
    Git.git.annotate: 0.000s total, mean: 0.000s, max: 0.000s
    SublimeCodeIntel.SublimeCodeIntel: 0.001s total, mean: 0.000s, max: 0.000s
    Deviot.Deviot: 0.004s total, mean: 0.000s, max: 0.001s
    GitGutter.modules.events: 0.001s total, mean: 0.000s, max: 0.000s
    SideBarEnhancements.SideBar: 0.001s total, mean: 0.000s, max: 0.000s
    SublimeLinter.sublimelinter: 0.001s total, mean: 0.000s, max: 0.000s
    Default.settings: 0.001s total, mean: 0.000s, max: 0.000s
    SideBarEnhancements.Stats: 0.000s total, mean: 0.000s, max: 0.000s

on_selection_modified:
    SublimeCodeIntel.SublimeCodeIntel: 0.002s total, mean: 0.000s, max: 0.001s
    SideBarEnhancements.Stats: 0.001s total, mean: 0.000s, max: 0.000s
    Vintage.vintage: 0.003s total, mean: 0.000s, max: 0.000s
```

I am running Gentoo with Sublime Build 3126 on the stable channel.

gepd, 2017-07-07:
I've been the last two days trying to install gentoo, but I always find a problem. The last problem was with the VMware image from osboxes, it runs well on my machine but I never finded the terminal or a way to install a package. 

So is it possible you can let me debug the plugin from your machine? with teamviewer or something similar?

m4rcu5, 2017-07-11:
Sorry, maybe I should have told that Gentoo is not the most user friendly distro. By default, it comes with nothing :)

I did some more testing. In the end it is a combination of plugins triggering this;
The Package Sync plugin scans the `Packages` folder on a regular basis, and as Deviot installs a complete `penv` environment there, this has not exponentially grown in size/amount of files. Causing the scan to use way more CPU power..
I mitigated it by excluding the Deviot folder.

One more thing I noticed during this test;
The Package Sync plugin syncs the Package configuration to my other machines (via Dropbox), but the Deviot preferences also contain the `installed` state and `last_check_update` state, which are local to the machine. This stops a PlatformIO installation on the other machines.

gepd, 2017-07-11:
Thanks for the report, I will think how to solve this, maybe a new file with the last_check_update information could help, so you can exclude only that file.

I'll let you know when this is available to test

gepd, 2017-07-15:
I've change the way to store the preferences, you can exclude the file located in `Packages/Users/Deviot/deviot.ini` 

Please install the develop branch and let me know if it's working for you.

gepd, 2017-08-14:
Open this issue again if you keep having problems

closed 2017-08-14 by gepd