gepd/Deviot #131: Changing languages is not working.

alexandrefernandesjs, 2017-07-06:
Before report an issue, please make sure it hasn't been reported before.

### Operating system
Linux checked so far.

### Description of problem
It is selected the change of the language to another language, but the change does not happen while Deviot is always in English.

### Expected Results
Solution of the problem so that all users can use in their preferred language.

### Additional info

*You can edit the text above according to your need*

gepd, 2017-07-06:
After change the language can you check if there is any error in the sublime console (`Main menu > View > Show console`)

What linux distro are you running?

alexandrefernandesjs, 2017-07-06:
###Console Output

[Deviot] begin ======================================================
reloading plugin Deviot.Deviot
[Deviot] reloading |-- Deviot
[Deviot] reloading |-- Deviot.Deviot
[Deviot] reloading | |-- Deviot.commands
reloading settings Packages/User/deviot.sublime-settings
[Deviot] reloading | | |-- Deviot.commands.deviot_new_sketch
[Deviot] reloading | | | |-- Deviot.libraries
[Deviot] reloading | | | |-- Deviot.libraries.paths
[Deviot] reloading | | | |-- Deviot.libraries.tools
[Deviot] reloading | | |-- Deviot.commands.deviot_select_boards
[Deviot] reloading | | | |-- Deviot.libraries.quick_menu
[Deviot] reloading | | | | |-- Deviot.platformio
[Deviot] reloading | | | | |-- Deviot.platformio.project_recognition
[Deviot] reloading | | | | |-- Deviot.libraries.quick_panel
[Deviot] reloading | | | | |-- Deviot.libraries.preferences_bridge
[Deviot] reloading | | | | | |-- Deviot.platformio.pio_bridge
[Deviot] reloading | | | | | | |-- Deviot.libraries.file
[Deviot] reloading | | | | | | |-- Deviot.platformio.command
[Deviot] reloading | | | | | |-- Deviot.libraries.configobj
[Deviot] reloading | | | | | |-- Deviot.libraries.configobj.configobj
[Deviot] reloading | | | | | | |-- Deviot.libraries.configobj.six
[Deviot] reloading | | | | | | |-- Deviot.libraries.configobj._version
[Deviot] reloading | | | | |-- Deviot.libraries.serial
[Deviot] reloading | | | | | |-- Deviot.libraries.pyserial
[Deviot] reloading | | | | | | |-- Deviot.libraries.pyserial.serialposix
[Deviot] reloading | | | | | | | |-- Deviot.libraries.pyserial.serialutil
[Deviot] reloading | | | | | |-- Deviot.libraries.pyserial.tools
[Deviot] reloading | | | | | |-- Deviot.libraries.pyserial.tools.list_ports
[Deviot] reloading | | | | | | |-- Deviot.libraries.pyserial.tools.list_ports_posix
[Deviot] reloading | | | | | |-- Deviot.libraries.messages
[Deviot] reloading | | | | | | |-- Deviot.libraries.I18n
[Deviot] reloading | | |-- Deviot.commands.deviot_select_environment
[Deviot] reloading | | |-- Deviot.commands.deviot_search_library
[Deviot] reloading | | | |-- Deviot.libraries.libraries
[Deviot] reloading | | | | |-- Deviot.libraries.thread_progress
[Deviot] reloading | | |-- Deviot.commands.deviot_remove_library
[Deviot] reloading | | |-- Deviot.commands.deviot_import_library
[Deviot] reloading | | | |-- Deviot.libraries.project_check
[Deviot] reloading | | |-- Deviot.commands.deviot_insert_library
[Deviot] reloading | | |-- Deviot.commands.deviot_libraries_examples
[Deviot] reloading | | |-- Deviot.commands.deviot_open_library_folder
[Deviot] reloading | | |-- Deviot.commands.deviot_rebuild_lib_list
[Deviot] reloading | | |-- Deviot.commands.deviot_extra_library_folder
[Deviot] reloading | | |-- Deviot.commands.deviot_remove_extra_library_folder
[Deviot] reloading | | |-- Deviot.commands.deviot_compile_sketch
[Deviot] reloading | | | |-- Deviot.platformio.compile
[Deviot] reloading | | | | |-- Deviot.platformio.initialize
[Deviot] reloading | | |-- Deviot.commands.deviot_upload_sketch
[Deviot] reloading | | | |-- Deviot.platformio.upload
[Deviot] reloading | | |-- Deviot.commands.deviot_overwrite_upload_baud
[Deviot] reloading | | |-- Deviot.commands.deviot_clean_sketch
[Deviot] reloading | | | |-- Deviot.platformio.clean
[Deviot] reloading | | |-- Deviot.commands.deviot_open_ini_file
[Deviot] reloading | | |-- Deviot.commands.deviot_show_console
[Deviot] reloading | | |-- Deviot.commands.deviot_hide_console
[Deviot] reloading | | |-- Deviot.commands.deviot_choose_programmer
[Deviot] reloading | | |-- Deviot.commands.deviot_show_terminal
[Deviot] reloading | | | |-- Deviot.platformio.pio_terminal
[Deviot] reloading | | |-- Deviot.commands.deviot_hide_terminal
[Deviot] reloading | | |-- Deviot.commands.deviot_select_port
[Deviot] reloading | | |-- Deviot.commands.deviot_set_password
[Deviot] reloading | | |-- Deviot.commands.deviot_toggle_serial_monitor
[Deviot] reloading | | |-- Deviot.commands.deviot_send_serial_monitor
[Deviot] reloading | | |-- Deviot.commands.deviot_output_console
[Deviot] reloading | | |-- Deviot.commands.deviot_automatic_scroll
[Deviot] reloading | | |-- Deviot.commands.deviot_auto_clean
[Deviot] reloading | | |-- Deviot.commands.deviot_choose_baudrate
[... 14 lines cut ...]
[Deviot] reloading | | |-- Deviot.commands.deviot_languages
[Deviot] reloading | | |-- Deviot.commands.deviot_donate
[Deviot] reloading | | |-- Deviot.commands.deviot_about
[Deviot] reloading | | |-- Deviot.commands.deviot_pio_about
[Deviot] reloading | | |-- Deviot.commands.deviot_clean_view
[Deviot] reloading | | |-- Deviot.commands.deviot_clean_console
[Deviot] reloading | | |-- Deviot.commands.deviot_reload
[Deviot] reloading | | | |-- Deviot.libraries.progress_bar
[Deviot] reloading | | | |-- Deviot.libraries.reloader
[Deviot] reloading | | | | |-- Deviot.libraries.stack_meter
[Deviot] reloading | | |-- Deviot.commands.deviot_set_ip
[Deviot] reloading | |-- Deviot.beginning
[Deviot] reloading | |-- Deviot.beginning.pio_install
[Deviot] reloading | | |-- Deviot.libraries.syntax
[Deviot] reloading |-- Deviot.libraries.top_menu
[Deviot] reloading | |-- Deviot.libraries.menu_files
[Deviot] installing dummy package
reloading plugin _dummy_package
reloading settings Packages/User/deviot.sublime-settings
[Deviot] removing dummy package
unloading plugin _dummy_package
[Deviot] end --------------------------------------------------------

###
Use Linux Mint

gepd, 2017-07-06:
can you check if restarting sublime after select the language solves the problem?

closed 2017-07-07 by alexandrefernandesjs