gepd/Deviot #132: Plugin globally overrides standard sublime hot keys

pokidovea, 2017-07-10:
```
 [alt+ctrl+p]
   prompt_select_workspace                  Default               
   deviot_upgrade_pio                       Deviot         
       
 [ctrl+shift+f]
   show_panel                               Default               
   deviot_search_library                    Deviot   

 [ctrl+shift+b]
   build                                    Default               
   deviot_select_boards                     Deviot   

 [alt+q]
   wrap_lines                               Default               
   show_overlay                             Deviot 

 [ctrl+shift+l]
   split_selection_into_lines               Default               
   deviot_clean_sketch                      Deviot  

 [ctrl+shift+m]
   expand_selection                         Default               
   deviot_toggle_serial_monitor             Deviot 
               
 [ctrl+shift+n]
   new_window                               Default               
   deviot_new_sketch                        Deviot     

 [ctrl+shift+s]
   prompt_save_as                           Default               
   deviot_select_port                       Deviot  

 [ctrl+shift+t]
   reopen_last_file                         Default               
   deviot_show_terminal                     Deviot  

 [ctrl+shift+w]
   close_window                             Default               
   deviot_set_password                      Deviot  
```
I can't use this hot keys in other projects. Also it doesn't allow to change this behaviour.

gepd, 2017-07-10:
What is you OS?

You can change the file located in `Packages/Deviot/shortcuts` and select the file corresponding to your OS.

I'd like to fix this for all users so I'll appreciate if you tell me you are in OSX or Linux

pokidovea, 2017-07-10:
Linux

gepd, 2017-07-11:
Thanks for your report, I've updated this in the last commit, It will be available in the next release, but if you want to install it now, you can use the develop branch. Please [read here](https://github.com/gepd/Deviot/wiki/Setup#developer)

closed 2017-08-14 by gepd