gepd/Deviot #133: Error using preinstalled PIO?

airdrummingfool, 2017-07-11:
### Operating system
macOS with PIO installed via homebrew:
```sh
$ pio --version
PlatformIO, version 3.4.0

$ which pio
/usr/local/bin/pio
```

### Description of problem
I just updated from a dev build of Deviot 2 (0d6cd78a14) to the current tip of the `develop` branch. I encountered some errors with the syntax language file (similar to #125) so I removed all Deviot settings, which seemed to force a reinstall.

When Sublime Text launched, I got the following message in the Deviot console:
```
[ Deviot 2.0.1 ] Setup
04:19:01 PlatformIO is installed in your system. You can enjoy Deviot now!
```

However, when I try to do anything, Deviot cannot seem to find PIO. This includes "Show Terminal":
```
[ Deviot ] PlatformIO Terminal

To get information about the PlatformIO command writes: 'pio -h'
or 'help' to know about the Deviot commands.

$ pio -h 
/usr/bin/python: No module named platformio
```

and "Upgrade PlatformIO":
```
[ Deviot 2.0.1 ] Starting...
04:22:29 Searching for PlatformIO Updates...

/usr/bin/python: No module named platformio
```

If I try to "use Development PIO version", installation seems to get stuck as soon as it tries to actually perform an (un)installation:
```
[ Deviot 2.0.1 ] Starting...
04:24:17 Uninstalling current version...
04:24:18 Installing 'develop' PlatformIO version...
Collecting https://github.com/platformio/platformio/archive/develop.zip
  Downloading https://github.com/platformio/platformio/archive/develop.zip (196kB)
Collecting arrow<1 (from platformio==3.4.1a5)
Collecting bottle<0.13 (from platformio==3.4.1a5)
Collecting click<6,>=5 (from platformio==3.4.1a5)
  Using cached click-5.1-py2.py3-none-any.whl
Collecting colorama (from platformio==3.4.1a5)
  Using cached colorama-0.3.9-py2.py3-none-any.whl
Collecting lockfile<0.13,>=0.9.1 (from platformio==3.4.1a5)
  Using cached lockfile-0.12.2-py2.py3-none-any.whl
Collecting pyserial!=3.3,<4,>=3 (from platformio==3.4.1a5)
  Using cached pyserial-3.2.1-py2.py3-none-any.whl
Collecting requests<3,>=2.4.0 (from platformio==3.4.1a5)
  Using cached requests-2.18.1-py2.py3-none-any.whl
Collecting semantic_version>=2.5.0 (from platformio==3.4.1a5)
Collecting python-dateutil (from arrow<1->platformio==3.4.1a5)
  Using cached python_dateutil-2.6.1-py2.py3-none-any.whl
Collecting urllib3<1.22,>=1.21.1 (from requests<3,>=2.4.0->platformio==3.4.1a5)
  Using cached urllib3-1.21.1-py2.py3-none-any.whl
Collecting idna<2.6,>=2.5 (from requests<3,>=2.4.0->platformio==3.4.1a5)
  Using cached idna-2.5-py2.py3-none-any.whl
Collecting certifi>=2017.4.17 (from requests<3,>=2.4.0->platformio==3.4.1a5)
  Using cached certifi-2017.4.17-py2.py3-none-any.whl
Collecting chardet<3.1.0,>=3.0.2 (from requests<3,>=2.4.0->platformio==3.4.1a5)
  Using cached chardet-3.0.4-py2.py3-none-any.whl
Collecting six>=1.5 (from python-dateutil->arrow<1->platformio==3.4.1a5)
  Using cached six-1.10.0-py2.py3-none-any.whl
Installing collected packages: six, python-dateutil, arrow, bottle, click, colorama, lockfile, pyserial, urllib3, idna, certifi, chardet, requests, semantic-version, platformio
  Found existing installation: six 1.4.1
    Uninstalling six-1.4.1:
```

After waiting a while, I tried unchecking "Use Development PIO Version", and Deviot tried to install the normal version, but again got stuck as soon as it started actually trying to install something:
```
[ Deviot 2.0.1 ] Starting...
04:26:47 Uninstalling current version...
04:26:48 Installing 'stable' PlatformIO version...
Collecting platformio
  Using cached platformio-3.4.0-py27-none-any.whl
Collecting lockfile<0.13,>=0.9.1 (from platformio)
  Using cached lockfile-0.12.2-py2.py3-none-any.whl
Collecting requests<3,>=2.4.0 (from platformio)
  Using cached requests-2.18.1-py2.py3-none-any.whl
Collecting bottle<0.13 (from platformio)
Collecting arrow<1 (from platformio)
Collecting click<6,>=5 (from platformio)
  Using cached click-5.1-py2.py3-none-any.whl
Collecting semantic-version>=2.5.0 (from platformio)
Collecting colorama (from platformio)
  Using cached colorama-0.3.9-py2.py3-none-any.whl
Collecting pyserial!=3.3,<4,>=3 (from platformio)
  Using cached pyserial-3.2.1-py2.py3-none-any.whl
Collecting urllib3<1.22,>=1.21.1 (from requests<3,>=2.4.0->platformio)
  Using cached urllib3-1.21.1-py2.py3-none-any.whl
Collecting idna<2.6,>=2.5 (from requests<3,>=2.4.0->platformio)
  Using cached idna-2.5-py2.py3-none-any.whl
Collecting certifi>=2017.4.17 (from requests<3,>=2.4.0->platformio)
  Using cached certifi-2017.4.17-py2.py3-none-any.whl
Collecting chardet<3.1.0,>=3.0.2 (from requests<3,>=2.4.0->platformio)
  Using cached chardet-3.0.4-py2.py3-none-any.whl
Collecting python-dateutil (from arrow<1->platformio)
  Using cached python_dateutil-2.6.1-py2.py3-none-any.whl
Collecting six>=1.5 (from python-dateutil->arrow<1->platformio)
  Using cached six-1.10.0-py2.py3-none-any.whl
Installing collected packages: lockfile, urllib3, idna, certifi, chardet, requests, bottle, six, python-dateutil, arrow, click, semantic-version, colorama, pyserial, platformio
```

gepd, 2017-07-12:
I've just installed homebrew in a macOS (10.10.5) and also platformio.

In my case, at the beginning it wasn't recognizing platformio, but I've fixed that in https://github.com/gepd/Deviot/commit/5bfd05b694122e3d8ee7cbe6ec65bae0bcf4f7ac

After that fix, I was able run all options without problem, so I couldn't replicate the issue.

The only think I 've made different is the installation of the plugin, instead of GIT I've use the package control as is indicated in [this page](https://github.com/gepd/Deviot/wiki/Setup#developer)

Can you try installing that way?

What is you current macOS version?

The syntax problem isn't directly related with Deviot but with PlatformIO, I'm working to implement a fix, PlatformIO doesn't like very well to work with `ino` files one solution is to change the extesion to `cpp`, but I know this could be a little annoying, so I'm search an alternative solution.

airdrummingfool, 2017-07-13:
I think my syntax problem was because the old version of Deviot provided an `Arduino.tmLanguage` file for syntax highlighting, but recently changed to providing the syntax highlight info via `libraries/syntax.py`. Once I figured out the issue it wasn't a big deal to update the syntax setting to `Deviot`.

My macOS version is `10.12.5`. I deleted the Deviot plugin from my plugins folder and removed all Deviot settings.

Once I start Sublime Text, Deviot gets reinstalled automatically (from Package Control), since it is in my installed packages list. The "Package Control Messages" screen comes up, as well as the Deviot console which prints
```
[ Deviot 2.0.1 ] Setup
02:18:13 PlatformIO is installed in your system. You can enjoy Deviot now!
```

Once I open the PlatformIO terminal, I have the same issues (and compiling still doesn't do anything):
```
[ Deviot ] PlatformIO Terminal

To get information about the PlatformIO command writes: 'pio -h'
or 'help' to know about the Deviot commands.

$ pio -h 
/usr/bin/python: No module named platformio

$ platformio -h 
/usr/bin/python: No module named platformio
```

gepd, 2017-07-14:
Ok I guess I've solved. I don't know why it was working in the machine I've tested (it shouldn't)

Please check if with the last commit (https://github.com/gepd/Deviot/commit/c3a276633a70657d6f49770770d56885a6cbd433) you keep having the issue

About the syntax problem, you're right, the old file should be removed, that will be my next commit if the main issue is solved

airdrummingfool, 2017-07-14:
@gepd that fixed it for me, thanks!

Regarding the old file - it *was* removed, which is what caused my issue. There appears to be no upgrade path between the old version and the new one - so any files I already had open and specified as `Arduino` syntax caused the error because the `Arduino.tmLanguage` file is gone in the new version. It's easy just to tell Sublime Text to use the `Deviot` syntax though, so it's not a big problem.

gepd, 2017-07-14:
yes, that is part of the plan. The plugin already loads the new syntax each time you open a file, but there is no action when the files are already loaded.

Thanks for the reports!

gepd, 2017-07-14:
Ok, I've fixed the syntax problem too.

closed 2017-07-14 by gepd