gepd/Deviot #134: Syntax highlighting gone when uninstalled Stino

nicmar, 2017-07-15:
I first installed Stino, and got syntax highlighting. Then I didn't like that, so I tried Deviot 2.0, which I love and want to use.

So I tried to get rid of the Arduino-menu that Stino created and I used Package Control > Uninstall package and removed Stino which was manually installed by placing the contents inside the Packages folder.

At that point, all the syntax highlighting was gone, and the Arduino-menu was still there. I tried installing Deviot again, no success.

How can I get Arduino-like syntax highlighting again? Is that a part of deviot, and how come I destroyed it?

Thanks!

nicmar, 2017-07-15:
Hmm, sorry, when I resaved the file I had opened, the colors came back. I still don't know how to get rid of the Arduino-menu from Stino though.

gepd, 2017-07-15:
To remove the `stino` menu go to `Packages\User` and remove the `stino` folder. That should remove immediately the menu

nicmar, 2017-07-15:
Hmm, it worked after I had removed the package for Stino again, but then the syntax highlighting is gone again. Isn't it supposed to be syntax highlighting in Deviot, or am I missing something?

If I delete the Stino-folder with Stino installed, it recreates on startup.

nicmar, 2017-07-15:
The only way I get syntax highlighting, is by having stino installed.

gepd, 2017-07-15:
First remove `stino` from the Package Control, after that remove the `stino` folder as I previously mentioned

If some file isn't highlighted, go to `ST Menu > View > Syntax` and select `Deviot` that should solve the problem, or you can open the file again

closed 2017-07-16 by nicmar