gepd/Deviot #135: Multiple Serial Send commands?

nicmar, 2017-07-15:
Is there a way to send multiple commands to the serial, without pressing Ctrl+Alt+S each time? So I can just input a command, press enter and input the next. If not, is there a way to change the shortcut, since I find it a little awkward :)

However, the new shortcuts in Deviot 2.0 are great, much better than ^K ^U etc to upload a sketch as in the old version.

Really love it :)

gepd, 2017-07-15:
I'm happy you love it!

In a previous version I've included a persistent option to avoid press short cut all the time, but I forgot about it when I wrote version 2.0. I've added it again. To try it, please install the `develop` branch reading [this instructions](https://github.com/gepd/Deviot/wiki/Setup#developer).

If you want to customise Deviot with your own shortcuts, make a new file called:

Windows:
`Default (Windows).sublime-keymap`
Linux
`Default (Linux).sublime-keymap`
OSX
`Default (OSX).sublime-keymap`

Store that file in `Packages\User\Deviot`

You can use the files in `Packages\Deviot\shortcuts` as reference

nicmar, 2017-07-16:
Great, thanks for your quick support!

nicmar, 2017-07-17:
Hi again, sorry for reopening, but it seems that all shortcut keys are different from the regular version. I'm just checking, will you be changing this in the main version? Upload used to be Ctrl+Shift+U, now it's ALT+E, and sometimes when I press that, i get "é" instead of something being uploaded, so I'm not sure the shortcuts work anymore..

Also, from time to time, I can't upload, and I need to close and reopen sublime for upload to work, but I think that was in the normal 2.0 version too.

gepd, 2017-07-17:
I have changed some of them because it was overridden some global shortcuts (issue: https://github.com/gepd/Deviot/issues/132)  you can propose a new ones or customize in your machine as I've previously explain. I'll check again that in a virtual machine.

You can't upload after got some errors in your code? I know there is a bug with the error linter and I'm working to fix that

nicmar, 2017-07-21:
Maybe, I'm not sure when the upload stops working, I'll try and see if it's something with after errors. Thanks for your great support, everyone should be using Deviot :)

gepd, 2017-07-27:
Sorry for the delay @nicmar 
you can check if the plugin is working (compiling/uploading) looking into the bar status, there is a progress message when one of that functions are working in the background

nicmar, 2017-07-27:
Hi, I bound uplod to the F5 key and I notice sometimes that I have to press twice before it uploads, even if there's no error before. 

It's no problem cause it's easy to just press F5 again. Choosing from the menu was more annoying. So this is really no issue for me, but it might still be a bug. 

Thanks!

gepd, 2017-07-27:
I'll check that, maybe there is something I can do. I'm still working in the fix related this problems so I hope when all be solved this issue goes away too

gepd, 2017-09-12:
It seems to be fixed with the last release, please reopen this issue if the problem keeps

closed 2017-09-12 by gepd