gepd/Deviot #137: Deviot corrupting platformio.ini

thinkyhead, 2017-07-17:
Before report an issue, please make sure it hasn't been reported before.

### Operating system

macOS 10.12.5

### Description of problem

Upon build, Deviot overwrites `platformio.ini` with the user's personal workspace data. It should not need to do this, and this workaround creates a nuisance and potential security risk for users of version control systems.

### Steps to Reproduce

Do a new install of Deviot and PlatformIO. Use `platformio.ini` in the manner allowed and prescribed by `platformio.ini` documentation for a project that doesn't have or want a `src` folder.

### Actual Results

The `platformio.ini` file is altered, leading to the possibility that unique local environment info will be checked in as a change to the global repository.

### Expected Results

The `platformio.ini` file is read, properly groked by Deviot, and left untouched.

gepd, 2017-07-17:
To avoid that, select `Options > Use PlatformIO Structure` in the Deviot Menu

fiveangle, 2017-07-27:
Identical issue here, but selecting `Options > Use PlatformIO` Structure did not resolve.  Windows 7, latest ST3/DevIoT and dev PIO (3.4.1a6)

gepd, 2017-07-27:
I've pushed a fix, please use the developer branch following [this instructions](https://github.com/gepd/Deviot/wiki/Setup#developer) and let me know if that solved the problem.

fiveangle, 2017-07-27:
That was an amazing easy switch to a realtime dev branch :+1: 

Fix appears to keep the bulk of the entries, but now removing the src_dir entry entirely:
[image]

```
[ Deviot 2.0.1 ] Starting...
[07/26/17 21:55:56] Processing Re-ARM (extra_scripts: Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py; build_flags: !python Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py; lib_deps: U8glib-ARM; board_f_cpu: 100000000L; platform: nxplpc; lib_ldf_mode: off; src_filter: +<*> -<example_configurations> -<src/HAL/HAL_*> +<src/HAL/HAL_LPC1768>)
Verbose mode can be enabled via `-v, --verbose` option
Collected 3 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <U8glib-ARM> v1.18.1
Error: Nothing to build. Please put your source code files to 'C:\dev\github\Marlin\src' folder
 [ERROR] Took 7.95 seconds 
 [SUMMARY] 
Environment megaatmega2560  	[SKIP]
Environment megaatmega1280  	[SKIP]
Environment printrboard     	[SKIP]
Environment printrboard_revf	[SKIP]
Environment brainwavepro    	[SKIP]
Environment rambo           	[SKIP]
Environment DUE             	[SKIP]
Environment teensy35        	[SKIP]
Environment Re-ARM          	[ERROR]
 [ERROR] Took 7.96 seconds 
```

gepd, 2017-07-27:
Ok, I've pushed new changes, please check and let me know if this time it work as you expect

gepd, 2017-08-14:
open this issue again if you keep having problems

fiveangle, 2017-08-14:
Sorry for not getting back to you.  It didn't resolve the problem but I questioned how to tell that I was in fact picking up the changes, and haven't had time to figure that out.

fiveangle, 2017-08-14:
@thinkyhead - are you still having this problem ? (I switched to Atom to try and get PIO working for the Re-ARM hw since DeVIoT appeared to not like the newer multi-line syntax for library inclusion)

gepd, 2017-08-14:
You keep getting the error or the changes didn't remove the `src_dir`?

fiveangle, 2017-08-14:
Changes did remove the src_dir entry, and so would get error that it couldn't find the source files.

gepd, 2017-08-14:
That is weird, I did replicate your issue the first time but after the last commit I don't got it anymore.

The new multi-line syntax should work too, but you should check if you have installed the last version of platformio

fiveangle, 2017-08-14:
since you're intersted in fixing, I'll try to prioritize this

gepd, 2017-08-14:
Thanks I appreciate that

If you still have installed deviot please try deselect `Options > Use PlatformIO Structure` since after the last changes I made, this behavior changed

fiveangle, 2017-08-16:
Opened [project](https://github.com/MarlinFirmware/Marlin/tree/2.0.x) `./platformio.ini` and `./Marlin/Marlin.ino`.  With ino tab active, attempted to select _Environment_ and option was greyed-out.  Edited platformio.ini to remove multi-line lib_deps and put on single comma-separated line and env was then available:

[image]

Ensured "Use platformio structure" unselected in Deviot

Selected Re-ARM, Compile, and it looks to have succeeded:

```
arm-none-eabi-objcopy -O binary .pioenvs/Re-ARM/firmware.elf .pioenvs/Re-ARM/firmware.bin
text	   data	    bss	    dec	    hex	filename
9844	    472	     44	  10360	   2878	.pioenvs/Re-ARM/firmware.elf
========================= [SUCCESS] Took 8.70 seconds =========================
================================== [SUMMARY] ==================================
Environment megaatmega2560  	[SKIP]
Environment megaatmega1280  	[SKIP]
Environment printrboard     	[SKIP]
Environment printrboard_revf	[SKIP]
Environment brainwavepro    	[SKIP]
Environment rambo           	[SKIP]
Environment DUE             	[SKIP]
Environment teensy35        	[SKIP]
Environment Re-ARM          	[SUCCESS]
========================= [SUCCESS] Took 8.70 seconds =========================
```

Reviewed changes to source, and found platformio.ini changed beyond the edit to consolidate the multi-line lib_deps, including removal of src filter for the selected _Re-ARM_ env:

[image]

ps- Let me know if you want me to open a new issue for the multi-line lib_deps issue.

gepd, 2017-08-16:
yes, would be good to open a new issue, so it will more easy to follow and better for future search from other users

fiveangle, 2017-08-26:
fyi... you probably already know this, but I just retested after resolution of the multi-line lib_deps issue and it seems to be moving backwards a bit, as it's now also strippinng comments :)

[image]

gepd, 2017-08-27:
I didn't know about that, I'll copy you coment to the other issue and will let it open it to fix that

fiveangle, 2017-09-05:
It seems the subject should be changed back to it's original:

> @gepd gepd changed the title from Deviot corrupting platformio.ini to platformio.ini untouched, without src_dir on Jul 27

gepd, 2017-09-05:
I don't think so, technically, the platformio file isn't corrupted, you can work with it, and the curruption word is more related to an error

fiveangle, 2017-09-21:
This appears to be fixed too :+1: 

Thanks @gepd !

gepd, 2017-09-21:
I didn't note it changed something when `Use PIO Structure` is selected, I'll check again to make it compatible as well

fiveangle, 2017-09-21:
I originally had _Use PIO Structure_ selected and got compile errors, but did not mean to indicate deviot changed the files.  I agree it would be good to confirm it doesn't just to be sure. (I can't check right now as not at my system).

fiveangle, 2017-09-24:
Okay, latest report (soooooo close):

When select "_Compile_", DevIoT will immediately duplicate first `lib_deps = ` line in a multi-line format lib_deps statement:

[image]

Selecting "_Use PIO Structure_" results in expected compile failure (cannot find source files) but `lib_deps =` line is duplicated same as when "_Use PIO Structure_" is not selected.  Good news is that the extra "lib_deps =" line does not cause compile to fail.

I do not know if this is related to the fix for this issue, #144, or entirely new.  Let me know if you would like me to open a separate issue, continue here, or re-open #144.

Thanks,

-=dave

gepd, 2017-09-24:
Fixed. You shouldn't have this problem in `v2.2.0-dev.5` and now it should have the same behavior with `Use PIO Structure` and without

fiveangle, 2017-09-25:
v2.2.0-dev.5 works like a charm in PIO structure mode and regular 👍

gepd, 2017-09-26:
I've added a new option in `Deviot Menu> Compile / Upload Options> platformio.ini untouch` to activate this feature, in the last commit I've solve the `lib_extra_dirs` problem too.

As reference, read [this issue](https://github.com/gepd/Deviot/issues/175) to know why was neccesary to add this option

It's available in `2.2.0-dev.6`

fiveangle, 2017-10-07:
Issue appears to be resolved

fiveangle, 2018-01-09:
As of 2.2.0 d12, **without** "_platformio.ini untouch_" option, the `platformio.ini` file appears to remain untouched.  As far as I know this can likely be closed for the orginal issue it was opened for.

However with the new "_platformio.ini untouch_" option enabled,  `platformio.ini` file ironically is very much touched, as it's entire content is deleted 😆 

-=dvae

thinkyhead, 2018-02-05:
I haven't been using Deviot for a few months while awaiting patches, but I will give the latest version a shot. Thanks for being diligent about these issues!

closed 2018-02-05 by thinkyhead

fiveangle, 2018-02-05:
@thinkyhead - you have to enable dev Deviot branch (currently rev12), plus enable Use development PIO, and don't enable "platformio.ini untouch" and it's been working like a charm.

thinkyhead, 2018-02-05:
Great! I really want to get back to building Marlin 2.0.x for the other platforms. I've got boards to flash!