gepd/Deviot #138: list available projects in build folder aka sketch folder.

bahhax404, 2017-07-25:
Thank you for the awesome plug-in, Arduino IDE is boring . 
is there a way to show the projects in a sub-menu for build folder .
something like the {Open sketch} in Stino plug-in . where all the existing projects are listed.
Gracias !

gepd, 2017-07-27:
I've not used the last version of stino, so I don't know very well how that feature is working; do you have to select previously what projects do you want in the menu or are these took from the list of projects from Sublime?

bahhax404, 2017-07-27:
It's similar to  the { Examples }   section where it displays examples from different libraries. 
what i'm talking about is a {Sketch} section in the menu that lists personal projects just like the above, without going to the projects folder or sublime menu. 
I'll check the way it's implemented in Stino and port it. 
another thing functions  are not highlighted with colors in Deviot syntax.  like analogRead() ...
thanks again for Deviot .

gepd, 2017-07-27:
I'll check more about that function and I'll let you know.

You mean only that kind of functions right? (analogRead, analogWrite...) 
but the rest of the code if colored?

bahhax404, 2017-07-27:
yes  functions only, keywords are highlighted. 
checked the stino syntax file they are defined like this :
  - match:\b(pinMode|digitalWrite|digitalRead|analogReference  .....
  -  scope: entity.name.function.arduino
in your Deviot syntax they don't get colored like keyword .

bahhax404, 2017-07-28:
I added the sections I was talking about, 
as lazy as i'm, I used the Stino libs and commands. I removed everything from Stino menu except the open sketch and open example ( although examples already exist in your menu as overlay ) .
I merged the two menus. 
- open sketch displays my projects that are in a folder which is the build folder too.
- examples are from the Arduino IDE folder. 
I like your plugin it's much faster and doesn't freeze when I use serial monitor in Sublime .

here is a screenshot :) 
[image]