gepd/Deviot #139: 2.0 slow opening files

tablatronix, 2017-08-09:
2.0 dev

```
  File "C:\Users\...\Documents\projects\bin\sublimetext\Data\Packages\Deviot\platformio\project_recognition.py", line 171, in get_file_extension
    extension = file_name.split(".")[1]
IndexError: list index out of range
```
I have this in my log a bit, when i turn off deviot, my files load fast.
filename is like `blah_blah$1.log`

gepd, 2017-08-09:
I've fixed the error in the console with the last commit, I do not know if that was causing the slow to open the file, if the issue continues, can you please give me more information? like 

what kind of file are you opening and what is the size?

gepd, 2017-09-12:
Do you keep having this problem with the last release?

tablatronix, 2017-09-12:
I have not noticed it again

gepd, 2017-09-12:
I'll close it for now, but if it come back, don't hesitate to open it again.

closed 2017-09-12 by gepd