gepd/Deviot #140: Deviot can't find platformio

ADeeds, 2017-08-11:
I'm having trouble with getting deviot up and running using ST3 and Mac OSX 10.11

When I installed Deviot it seemed to be going OK

> [ Deviot 2.0.1 ] Setup
22:03:12 PlatformIO is installed in your system. You can enjoy Deviot now!

However I can't seem to get any of the Deviot menu options to do anything. I'm assuming something should pop up when I try to do actions like "Select Board"

When I go to Options-> Upgrade PlatformIO, I get this message:

> [ Deviot 2.0.1 ] Starting...
22:07:51 Searching for PlatformIO Updates...

/usr/bin/python: No module named platformio

gepd, 2017-08-12:
Are you using the master or develop branch?

If you don't know probably is the master, so please follow [this instructions](https://github.com/gepd/Deviot/wiki/Setup#developer) after install the new branch restart ST and let me know if you keep with the problem

ADeeds, 2017-08-12:
Changing the branch that Package Control is getting seems to have helped partially.

When I hit Upgrade, it seems to work
> [ Deviot 2.0.1 ] Starting...
00:50:29 Searching for PlatformIO Updates...
You're up-to-date!
PlatformIO 3.4.1 is currently the newest version available.

However still nothing happens when I choose "Select Board" from the menu. Here's the error that's being written to the Sublime console

>Traceback (most recent call last):
  File "./python3.3/json/decoder.py", line 367, in raw_decode
StopIteration
\
During handling of the above exception, another exception occurred:
\
Traceback (most recent call last):
  File "/Applications/Sublime Text.app/Contents/MacOS/sublime_plugin.py", line 797, in run_
    return self.run()
  File "/Users/<username>/Library/Application Support/Sublime Text 3/Packages/Deviot/commands/deviot_select_boards.py", line 6, in run
    QuickMenu().quick_boards()
  File "/Users/<username>/Library/Application Support/Sublime Text 3/Packages/Deviot/libraries/quick_menu.py", line 34, in quick_boards
    boards_list = self.quick_boards_list()
  File "/Users/<username>/Library/Application Support/Sublime Text 3/Packages/Deviot/libraries/quick_menu.py", line 77, in quick_boards_list
    boards = boards_file.read_json()
  File "/Users/<username>/Library/Application Support/Sublime Text 3/Packages/Deviot/libraries/file.py", line 57, in read_json
    return json.loads(file)
  File "./python3.3/json/__init__.py", line 316, in loads
  File "./python3.3/json/decoder.py", line 351, in decode
  File "./python3.3/json/decoder.py", line 369, in raw_decode
ValueError: No JSON object could be decoded

gepd, 2017-08-12:
Go to `Deviot Menu > Options > Rebuild Boards File` it will take 2-3 seconds to rebuild the corrupted file, now you should be able to see the list of boards

ADeeds, 2017-08-12:
Thanks! That fixed it

closed 2017-08-12 by ADeeds

erm3nda, 2018-11-01:
I had the same problem and i managed to solve using the option "use dev version of IOT" from menu, then rebuild board list. Upgrading to dev version itself didn't solve anything.

:-)