gepd/Deviot #141: Serial COM port not listed Win7_64

aj2004, 2017-08-12:
I'm running Windows 7 64-bit with the latest Deviot 2.0.1.
I have an Arduino Mega (2560) on USB/COM6, but it's not listed when I select "Serial Ports(s)".
Device Manager shows the Arduino connected and installed okay. Uploading works from the Arduino IDE.
It was working as of a day or 2 ago and I don't remember installing anything that should conflict with this.

I have the same versions of ST and Deviot installed on my laptop running Ubuntu and it is working fine.

The only output in the console I get is "Select a port from the list".

I've tried removing Deviot and reinstalling, and I've tried reinstalling ST with no luck.

gepd, 2017-08-13:
Please check the ST console (`View > Show Console`) after select the list of ports, if there is an error please paste here

Some days ago I've made some small changes in this area, please follow [this instructions](https://github.com/gepd/Deviot/wiki/Setup#developer) to install the develop branch and let me know if this fix the problem

Try that two things and if the error keep after change to the develop branch try to look the ST console too to see where could be the problem

aj2004, 2017-08-13:
When starting ST, this is the only error message I get from the console (after plugins have loaded):

`plugins loaded`
`Traceback (most recent call last):`
`  File "Z:\Program Files\Sublime Text 3\sublime_plugin.py", line 210, in on_api_ready`
`    m.plugin_loaded()`
`  File "C:\Users\kevin\AppData\Roaming\Sublime Text 3\Packages\Deviot\Deviot.py", line 46, in` `plugin_loaded`
`    active_window().run_command('deviot_rebuild_syntax')`
`NameError: global name 'active_window' is not defined`
`reloading settings Packages/User/deviot.sublime-settings`

No message is given when attempting to select a serial port. The options I have are:
- Add IP Manually
- Communications Port (COM1) (which is the parallel port)

I tried the develop branch and get the same result.

gepd, 2017-08-13:
What is the version of your ST?
`active_window` is a function from ST and should be recognized, I'll try to replicate it in a virtual machine

aj2004, 2017-08-14:
Both Windows (not working) and Linux (working) are the same version 3, build 3126.
I checked the console on Linux and do not get the errors.

gepd, 2017-08-14:
Can you list what others plugins have you installed in your machine?
I could not replicate the problem

aj2004, 2017-08-14:
Default.auto_indent_tag
Default.block
Default.comment
Default.convert_syntax
Default.copy_path
Default.delete_word
Default.detect_indentation
Default.duplicate_line
Default.echo
Default.exec
Default.fold
Default.font
Default.goto_line
Default.history_list
Default.indentation
Default.install_package_control
Default.kill_ring
Default.mark
Default.new_templates
Default.open_context_url
Default.open_in_browser
Default.pane
Default.paragraph
Default.paste_from_history
Default.profile
Default.quick_panel
Default.run_syntax_tests
Default.save_on_focus_lost
Default.scroll
Default.set_unsaved_view_name
Default.settings
Default.show_scope_name
Default.side_bar
Default.sort
Default.swap_line
Default.switch_file
Default.symbol
Default.transform
Default.transpose
Default.trim_trailing_white_space
CSS.css_completions
Diff.diff
HTML.encode_html_entities
HTML.html_completions
0_package_control_loader.00-package_control
Package Control.1_reloader
Package Control.2_bootstrap
Package Control.Package Control
Verbose.Verbose
Deviot.Deviot

I have another computer with Win10 64-bit I will try tomorrow.
Is there a way to manually specify the COM port? I tried adding `upload_port = COM6` to the build config file `platformio.ini` but the setting wasn't used when I tried to upload.

gepd, 2017-08-21:
Can you check if you keep having problem with the last changes I made?

aj2004, 2017-08-23:
No change.  I have ST installed on a USB drive (portable) and it works on another Windows computer (but not this one), so I'm lead to believe that there is just something not working with my computer's setup rather than an issue with your package.  The "Deviot" menu bar sometimes shows and sometimes doesn't.  The same with the Deviot syntax selection (bottom-right).

gepd, 2017-08-23:
I've pushed some changes in the syntax feature in the last commits, I didn't test them but I'll check again to see where is the problem.

Did you noted if the error `NameError: global name 'active_window' is not defined` was fixed?

aj2004, 2017-08-23:
It's still there

gepd, 2017-08-23:
Do you have installed the developer branch? 

[this](https://github.com/gepd/Deviot/commit/edd363ac2e2f496af4459ae409110c5440f7f0ca) commit should be resolved that issue

aj2004, 2017-08-23:
With the developer branch, I have the following errors:

>Traceback (most recent call last):
  >File "D:\apps\sublime_text64\sublime_plugin.py", line 210, in on_api_ready
    >m.plugin_loaded()
  >File "D:\apps\sublime_text64\Data\Packages\Deviot\Deviot.py", line 29, in plugin_loaded
    >PioInstall()
  >File "D:\apps\sublime_text64\Data\Packages\Deviot\beginning\pio_install.py", line 64, in __init__
    >installed = get_sysetting('installed', False)
  >File "D:\apps\sublime_text64\Data\Packages\Deviot\libraries\tools.py", line 121, in get_sysetting
    >config.read(sys_path)
  >File "D:\apps\sublime_text64\Data\Packages\Deviot\libraries\configparser\__init__.py", line 705, in read
    >self._read(fp, filename)
  >File "D:\apps\sublime_text64\Data\Packages\Deviot\libraries\configparser\__init__.py", line 1101, in _read
    >raise MissingSectionHeaderError(fpname, lineno, line)
>Deviot.libraries.configparser.MissingSectionHeaderError: File contains no section headers.
>file: 'D:\\apps\\sublime_text64\\Data\\Packages\\User\\Deviot\\deviot.ini', line: 1
'env_path = D:\\apps\\sublime_text64\\Data\\Packages\\User\\Deviot\\penv;D:\\apps\\sublime_text64\\Data\\Packages\\User\\Deviot\\penv\\Scripts;C:\\Python27\\;C:\\Python27\\Scripts;C:\\WINDOWS\\system32;C:\\WINDOWS;C:\\WINDOWS\\System32\\Wbem;C:\\WINDOWS\\System32\\Wi>ndowsPowerShell\\v1.0\\;C:\\Program Files (x86)\\ATI Technologies\\ATI.ACE\\Core-Static;C:\\Program Files (x86)\\AMD\\ATI.ACE\\Core-Static;C:\\Users\\Pacificon\\AppData\\Local\\Microsoft\\WindowsApps;\n'
>Traceback (most recent call last):
  >File "D:\apps\sublime_text64\sublime_plugin.py", line 389, in run_callback
    >expr()
  >File "D:\apps\sublime_text64\sublime_plugin.py", line 522, in <lambda>
    >run_callback('on_activated', callback, lambda: callback.on_activated(v))
  >File "D:\apps\sublime_text64\Data\Packages\Deviot\Deviot.py", line 50, in on_activated
    >PreferencesBridge().set_status_information()
  >File "D:\apps\sublime_text64\Data\Packages\Deviot\libraries\preferences_bridge.py", line 19, in __init__
    >super(PreferencesBridge, self).__init__()
  >File "D:\apps\sublime_text64\Data\Packages\Deviot\platformio\pio_bridge.py", line 19, in __init__
    >super(PioBridge, self).__init__()
  >File "D:\apps\sublime_text64\Data\Packages\Deviot\platformio\command.py", line 25, in __init__
    >env_path = get_sysetting('env_path', None)
  >File "D:\apps\sublime_text64\Data\Packages\Deviot\libraries\tools.py", line 121, in get_sysetting
    >config.read(sys_path)
  >File "D:\apps\sublime_text64\Data\Packages\Deviot\libraries\configparser\__init__.py", line 705, in read
    >self._read(fp, filename)
  >File "D:\apps\sublime_text64\Data\Packages\Deviot\libraries\configparser\__init__.py", line 1101, in _read
    >raise MissingSectionHeaderError(fpname, lineno, line)
>Deviot.libraries.configparser.MissingSectionHeaderError: File contains no section headers.
>file: 'D:\\apps\\sublime_text64\\Data\\Packages\\User\\Deviot\\deviot.ini', line: 1
'env_path = D:\\apps\\sublime_text64\\Data\\Packages\\User\\Deviot\\penv;D:\\apps\\sublime_text64\\Data\\Packages\\User\\Deviot\\penv\\Scripts;C:\\Python27\\;C:\\Python27\\Scripts;C:\\WINDOWS\\system32;C:\\WINDOWS;C:\\WINDOWS\\System32\\Wbem;C:\\WINDOWS\\System32\\Wi>ndowsPowerShell\\v1.0\\;C:\\Program Files (x86)\\ATI Technologies\\ATI.ACE\\Core-Static;C:\\Program Files (x86)\\AMD\\ATI.ACE\\Core-Static;C:\\Users\\Pacificon\\AppData\\Local\\Microsoft\\WindowsApps;\n'
>Traceback (most recent call last):
  >File "D:\apps\sublime_text64\sublime_plugin.py", line 389, in run_callback
    >expr()
  >File "D:\apps\sublime_text64\sublime_plugin.py", line 522, in <lambda>
    >run_callback('on_activated', callback, lambda: callback.on_activated(v))
  >File "D:\apps\sublime_text64\Data\Packages\Deviot\Deviot.py", line 50, in on_activated
    >PreferencesBridge().set_status_information()
  >File "D:\apps\sublime_text64\Data\Packages\Deviot\libraries\preferences_bridge.py", line 19, in __init__
    >super(PreferencesBridge, self).__init__()
  >File "D:\apps\sublime_text64\Data\Packages\Deviot\platformio\pio_bridge.py", line 19, in __init__
    >super(PioBridge, self).__init__()
  >File "D:\apps\sublime_text64\Data\Packages\Deviot\platformio\command.py", line 25, in __init__
    >env_path = get_sysetting('env_path', None)
  >File "D:\apps\sublime_text64\Data\Packages\Deviot\libraries\tools.py", line 121, in get_sysetting
    >config.read(sys_path)
  >File "D:\apps\sublime_text64\Data\Packages\Deviot\libraries\configparser\__init__.py", line 705, in read
    >self._read(fp, filename)
  >File "D:\apps\sublime_text64\Data\Packages\Deviot\libraries\configparser\__init__.py", line 1101, in _read
    >raise MissingSectionHeaderError(fpname, lineno, line)
>Deviot.libraries.configparser.MissingSectionHeaderError: File contains no section headers.
>file: 'D:\\apps\\sublime_text64\\Data\\Packages\\User\\Deviot\\deviot.ini', line: 1

And the Deviot menu bar item is not shown. However, the Deviot syntax option is there.
This is running ST portable

gepd, 2017-08-23:
Fixed, this was an issue related with the last changed made for https://github.com/gepd/Deviot/issues/144

closed 2017-09-11 by gepd