gepd/Deviot #143: Compilacion

alito56, 2017-08-18:
Before report an issue, please make sure it hasn't been reported before.

### Operating system Linux Sonya

### Board used arduino uno

### Description of problem:   imposible de compilar, me da el siguiente mensage "/bin/sh: 1: /home/alito/.config/sublime-text-3/Packages/User/Deviot/penv/bin/platformio: not found"

### Steps to Reproduce

### Actual Results

### Expected Results

### Additional info

*You can edit the text above according to your need*

gepd, 2017-08-18:
Cuando instalaste el plugin desde el package control, comenzó el proceso de instalación de las dependencias (platformio)?

alito56, 2017-08-19:
No, no istala nada,y esto lo hice al menos 10 veces

gepd, 2017-08-19:
Puedes intentarlo una vez más, desinstala el plugin previamente, luego comienza la instalación desde package control y una vez que hagas eso abre la consola de sublime text `View > Show Console` y me dices si aparece algún error o en caso de que no sepas, pega lo que hay en esa ventana, aquí

alito56, 2017-08-19:
Hice lo que pediste, aqui el resultado:

DPI scale: 1
startup, version: 3126 linux x64 channel: stable
executable: /opt/sublime_text/sublime_text
working dir: /home/alito
packages path: /home/alito/.config/sublime-text-3/Packages
state path: /home/alito/.config/sublime-text-3/Local
zip path: /opt/sublime_text/Packages
zip path: /home/alito/.config/sublime-text-3/Installed Packages
ignored_packages: ["Vintage"]
pre session restore time: 0.399296
startup time: 0.918635
first paint time: 0.947329
first paint time: 1.02133
reloading plugin Default.auto_indent_tag
reloading plugin Default.block
reloading plugin Default.comment
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.install_package_control
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.quick_panel
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.settings
reloading plugin Default.show_scope_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin 0_package_control_loader.00-package_control
reloading plugin 1Self.1self
reloading plugin 1Self.1selfOpenDashboardCommand
reloading plugin Alignment.Alignment
reloading plugin Emmet.emmet-plugin
reloading plugin Package Control.1_reloader
reloading plugin Package Control.2_bootstrap
reloading plugin Package Control.Package Control
reloading plugin SideBarEnhancements.SideBar
reloading plugin SideBarEnhancements.SideBarAPI
reloading plugin SideBarEnhancements.SideBarDefaultDisable
reloading plugin arduino-cli.arduino-cli
reloading plugin Anaconda.anaconda
reloading plugin Anaconda.setup
reloading plugin Anaconda.version
reloading plugin ColorPicker.sublimecp
plugins loaded
Initializing 1self plugin
Traceback (most recent call last):
   File "./python3.3/urllib/request.py", line 1248, in do_open
   File "./python3.3/http/client.py", line 1065, in request
   File "./python3.3/http/client.py", line 1103, in _send_request
   File "./python3.3/http/client.py", line 1061, in endheaders
   File "./python3.3/http/client.py", line 906, in _send_output
   File "./python3.3/http/client.py", line 844, in send
   File "./python3.3/http/client.py", line 822, in connect
   File "./python3.3/socket.py", line 417, in create_connection
socket.gaierror: [Errno -2] Name or service not known

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
   File "/opt/sublime_text/sublime_plugin.py", line 210, in on_api_ready
     m.plugin_loaded()
   File "1self in /home/alito/.config/sublime-text-3/Installed 
Packages/1Self.sublime-package", line 58, in plugin_loaded
   File "1self in /home/alito/.config/sublime-text-3/Installed 
Packages/1Self.sublime-package", line 65, in after_loaded
   File "1self in /home/alito/.config/sublime-text-3/Installed 
Packages/1Self.sublime-package", line 135, in get_stream_id_if_not_present
   File "./python3.3/urllib/request.py", line 156, in urlopen
   File "./python3.3/urllib/request.py", line 469, in open
   File "./python3.3/urllib/request.py", line 487, in _open
   File "./python3.3/urllib/request.py", line 447, in _call_chain
   File "./python3.3/urllib/request.py", line 1274, in http_open
   File "./python3.3/urllib/request.py", line 1251, in do_open
urllib.error.URLError: <urlopen error [Errno -2] Name or service not known>
Emmet: No need to update PyV8
Package Control: Skipping automatic upgrade, last run at 2017-08-19 
18:20:41, next run at 2017-08-19 19:20:41 or after
ignored packages updated to: ["Deviot", "Vintage"]
reloading settings Packages/User/Package Control.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings
reloading settings Packages/User/Package Control.sublime-settings
ignored packages updated to: ["Vintage"]
reloading plugin Deviot.Deviot
Traceback (most recent call last):
   File "/opt/sublime_text/sublime_plugin.py", line 157, in reload_plugin
     m.plugin_loaded()
   File "/home/alito/.config/sublime-text-3/Packages/Deviot/Deviot.py", 
line 46, in plugin_loaded
     active_window().run_command('deviot_rebuild_syntax')
NameError: global name 'active_window' is not defined
generating syntax summary
generating meta info summary
reloading settings Packages/User/Package Control.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings
reloading settings Packages/User/deviot.sublime-settings
reloading settings Packages/User/Preferences.sublime-settings

gepd, 2017-08-19:
Gracias por la información, he estado buscando una solución y al parecer, existe un problema con tu máquina y algún servidor dns, o algo así.

Por favor revisa la siguiente URL y dime si puedes acceder a ella (desde la misma maquina que estás realizando la instalación):

[https://pypi.python.org/packages/source/v/virtualenv/virtualenv-14.0.6.tar.gz](https://pypi.python.org/packages/source/v/virtualenv/virtualenv-14.0.6.tar.gz)

(Es un archivo, no es necesario que lo descargues)

Hay otra solución que sería instalar PlatformIO de forma manual, si tienes pip instalado puedes ejecutar el siguiente comando desde la terminal (si no tienes pip, [aquí](https://www.tecmint.com/install-pip-in-linux/) hay una guía, tiene que ser para python2)

`pip install -U platformio`

Una vez instalado, cuando abras Sublime Text debería reconocerlo automáticamente

alito56, 2017-08-19:
No hay problema, pude acceder a la pagina, te cuento que trate de 
instalar platformio manualmente

pero me dice lo siguiente:

Collecting platformio
   Using cached platformio-3.4.1.tar.gz
     Complete output from command python setup.py egg_info:
     PlatformIO Core v3.4.1 does not run under Python version 3.5.2.
     Minimum supported version is 2.7, please upgrade Python.
     Python 3 is not yet supported.

No entiendo pue tengo las dos versiones de python 2.7 y 3.5

gepd, 2017-08-19:
y con qué comando estás instalado platformio? 

Puede ser que `pip` esté vinculado a python 3 y `pip2` a python 2. PlatformIO solo funciona con python2 por ahora, asi es que solo tendrías que verificar que estés instalándolo desde el comando correcto

alito56, 2017-08-19:
esto es lo que me da:

Jupiter alito # pip2 install -U platformio
Requirement already up-to-date: platformio in 
/usr/local/lib/python2.7/dist-packages
Cleaning up...

gepd, 2017-08-19:
Ya lo tienes instalado, cuando ejecutas 

`platformio --version`

¿qué te aparece?

alito56, 2017-08-19:
Aqui tienes:

alito@Jupiter ~ $ platformio --version
PlatformIO, version 3.4.1

gepd, 2017-08-20:
cuando abres Sublime Text aparece algún mensaje como: "Instalando Deviot"?
Tu S.O. es Linux Mint Sonya?

Estoy intentando replicar el problema en una máquina virtual, hasta ahora las pruebas que hice en ambos casos funciona; instalando PlatformIO automáticamente y manualmente. 

Puedes contactarme en gepd[arroba]outlook.com así lo puedo revisar por teamviewer (si quieres)

closed 2017-09-12 by gepd