gepd/Deviot #144: Doesn't recognize multi-line lib_deps entry

fiveangle, 2017-08-18:
Mac OS Siera, Windows 7

Doesn't recognize multi-line lib_deps entry in _common_ section of _platformio.ini_

* Open project with multi-line lib_deps like [this one](https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x)
* Attempt to select _Environment_ and observe unable to do so
* Edit multi-line lib_deps entry to be single-line comma-separated:
[image]
* Observe able to select _Environment_ now

gepd, 2017-08-21:
I've changed the library used to open .ini files, now, it seems to work fine with multi-line options, please test it and let me know if it's working for you too

fiveangle, 2017-08-26:
Did a package upgrade/overwrite and with original multiline source, it picked up the libraries no problem.

Thanks !

-=dave

```
[ Deviot 2.1.1 ] Starting...
[Fri Aug 25 20:19:06 2017] Processing Re-ARM (extra_scripts: Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py; build_flags: !python Marlin/src/HAL/HAL_LPC1768/lpc1768_flag_script.py; lib_deps: U8glib-ARM; board_f_cpu: 100000000L; platform: nxplpc; lib_ldf_mode: off; src_filter: +<*> -<example_configurations> -<src/HAL/HAL_*> -<frameworks> +<src/HAL/HAL_LPC1768> +<frameworks/CMSIS/LPC1768> -</Users/speedster/dev/github/Marlin/Marlin/Marlin.ino> +</Users/speedster/dev/github/Marlin/Marlin/Marlin.cpp>)
--------------------------------------------------------------------------------
Library Storage: /Users/speedster/dev/github/Marlin/.piolibdeps
LibraryManager: Installing id=1691
U8glib-ARM @ 1.18.1 is already installed
Collected 8 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <U8glib-ARM> v1.18.1 (/Users/speedster/dev/github/Marlin/.piolibdeps/U8glib-ARM_ID1691)
arm-none-eabi-g++ -o .pioenvs/Re-ARM/src/G26_Mesh_Validation_Tool.o -c -fno-rtti -fno-exceptions -std=gnu++11 -fabi-version=0 -fno-use-cxa-atexit -fno-threadsafe-statics -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -nostdlib -Os -mcpu=cortex-m3 -mthumb -ffreestanding -fsigned-char -fno-move-loop-invariants -Wall --specs=nano.specs --specs=nosys.specs -MMD -MP -DF_CPU=100000000L -DPLATFORMIO=30500 -DTARGET_LPC1768 -DIS_REARM -IMarlin/frameworks/CMSIS/LPC1768/include -IMarlin/frameworks/CMSIS/LPC1768/lib -IMarlin/src/HAL -IMarlin -I.piolibdeps/U8glib-ARM_ID1691/src Marlin/G26_Mesh_Validation_Tool.cpp
arm-none-eabi-g++ -o .pioenvs/Re-ARM/src/I2CPositionEncoder.o -c -fno-rtti -fno-exceptions -std=gnu++11 -fabi-version=0 -fno-use-cxa-atexit -fno-threadsafe-statics -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -nostdlib -Os -mcpu=cortex-m3 -mthumb -ffreestanding -fsigned-char -fno-move-loop-invariants -Wall --specs=nano.specs --specs=nosys.specs -MMD -MP -DF_CPU=100000000L -DPLATFORMIO=30500 -DTARGET_LPC1768 -DIS_REARM -IMarlin/frameworks/CMSIS/LPC1768/include -IMarlin/frameworks/CMSIS/LPC1768/lib -IMarlin/src/HAL -IMarlin -I.piolibdeps/U8glib-ARM_ID1691/src Marlin/I2CPositionEncoder.cpp
arm-none-eabi-g++ -o .pioenvs/Re-ARM/src/M100_Free_Mem_Chk.o -c -fno-rtti -fno-exceptions -std=gnu++11 -fabi-version=0 -fno-use-cxa-atexit -fno-threadsafe-statics -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -nostdlib -Os -mcpu=cortex-m3 -mthumb -ffreestanding -fsigned-char -fno-move-loop-invariants -Wall --specs=nano.specs --specs=nosys.specs -MMD -MP -DF_CPU=100000000L -DPLATFORMIO=30500 -DTARGET_LPC1768 -DIS_REARM -IMarlin/frameworks/CMSIS/LPC1768/include -IMarlin/frameworks/CMSIS/LPC1768/lib -IMarlin/src/HAL -IMarlin -I.piolibdeps/U8glib-ARM_ID1691/src Marlin/M100_Free_Mem_Chk.cpp
arm-none-eabi-g++ -o .pioenvs/Re-ARM/src/Marlin.o -c -fno-rtti -fno-exceptions -std=gnu++11 -fabi-version=0 -fno-use-cxa-atexit -fno-threadsafe-statics -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -nostdlib -Os -mcpu=cortex-m3 -mthumb -ffreestanding -fsigned-char -fno-move-loop-invariants -Wall --specs=nano.specs --specs=nosys.specs -MMD -MP -DF_CPU=100000000L -DPLATFORMIO=30500 -DTARGET_LPC1768 -DIS_REARM -IMarlin/frameworks/CMSIS/LPC1768/include -IMarlin/frameworks/CMSIS/LPC1768/lib -IMarlin/src/HAL -IMarlin -I.piolibdeps/U8glib-ARM_ID1691/src Marlin/Marlin.cpp
Marlin/Marlin.ino:0:21: fatal error: Arduino.h: No such file or directory
#include <Arduino.h>
^
compilation terminated.
arm-none-eabi-g++ -o .pioenvs/Re-ARM/src/Marlin_main.o -c -fno-rtti -fno-exceptions -std=gnu++11 -fabi-version=0 -fno-use-cxa-atexit -fno-threadsafe-statics -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -nostdlib -Os -mcpu=cortex-m3 -mthumb -ffreestanding -fsigned-char -fno-move-loop-invariants -Wall --specs=nano.specs --specs=nosys.specs -MMD -MP -DF_CPU=100000000L -DPLATFORMIO=30500 -DTARGET_LPC1768 -DIS_REARM -IMarlin/frameworks/CMSIS/LPC1768/include -IMarlin/frameworks/CMSIS/LPC1768/lib -IMarlin/src/HAL -IMarlin -I.piolibdeps/U8glib-ARM_ID1691/src Marlin/Marlin_main.cpp
arm-none-eabi-g++ -o .pioenvs/Re-ARM/src/Sd2Card.o -c -fno-rtti -fno-exceptions -std=gnu++11 -fabi-version=0 -fno-use-cxa-atexit -fno-threadsafe-statics -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -nostdlib -Os -mcpu=cortex-m3 -mthumb -ffreestanding -fsigned-char -fno-move-loop-invariants -Wall --specs=nano.specs --specs=nosys.specs -MMD -MP -DF_CPU=100000000L -DPLATFORMIO=30500 -DTARGET_LPC1768 -DIS_REARM -IMarlin/frameworks/CMSIS/LPC1768/include -IMarlin/frameworks/CMSIS/LPC1768/lib -IMarlin/src/HAL -IMarlin -I.piolibdeps/U8glib-ARM_ID1691/src Marlin/Sd2Card.cpp
arm-none-eabi-g++ -o .pioenvs/Re-ARM/src/SdBaseFile.o -c -fno-rtti -fno-exceptions -std=gnu++11 -fabi-version=0 -fno-use-cxa-atexit -fno-threadsafe-statics -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -nostdlib -Os -mcpu=cortex-m3 -mthumb -ffreestanding -fsigned-char -fno-move-loop-invariants -Wall --specs=nano.specs --specs=nosys.specs -MMD -MP -DF_CPU=100000000L -DPLATFORMIO=30500 -DTARGET_LPC1768 -DIS_REARM -IMarlin/frameworks/CMSIS/LPC1768/include -IMarlin/frameworks/CMSIS/LPC1768/lib -IMarlin/src/HAL -IMarlin -I.piolibdeps/U8glib-ARM_ID1691/src Marlin/SdBaseFile.cpp
*** [.pioenvs/Re-ARM/src/Marlin.o] Error 1
[... cut ...]
========================== [ERROR] Took 5.76 seconds ==========================
================================== [SUMMARY] ==================================
Environment megaatmega2560  	[SKIP]
Environment megaatmega1280  	[SKIP]
Environment printrboard     	[SKIP]
Environment printrboard_revf	[SKIP]
Environment brainwavepro    	[SKIP]
Environment rambo           	[SKIP]
Environment DUE             	[SKIP]
Environment teensy35        	[SKIP]
Environment Re-ARM          	[ERROR]
========================== [ERROR] Took 5.76 seconds ==========================

```

gepd, 2017-08-27:
>fyi... you probably already know this, but I just retested after resolution of the multi-line lib_deps issue and it seems to be moving backwards a bit, as it's now also strippinng comments and swashing multi blank/trailing lines :)

[image]

fiveangle, 2017-09-05:
the multi-line library issue I reported here is resolved, the platformio.ini corruption issue of #137 remains and has gotten worse, so I'm unsure why you want to track that here rather than on #137

gepd, 2017-09-05:
Is easier to me, track one topic in a issue at time, than many of them in one.
In this case, the new problem started when I changed the library for this issue, so it's directly related. 
I'm currently working in this, because it will solve other related problems too, I think it will be done this week

fiveangle, 2017-09-05:
gotcha, thanks !

gepd, 2017-09-17:
I had forgot to tell you that this problem is fixed, you can tried installing the developer branch: https://github.com/gepd/Deviot/wiki/Setup#developer

fiveangle, 2017-09-21:
Appears resolved - thanks !

closed 2017-09-21 by fiveangle