gepd/Deviot #145: no JSON could be decoded

lexxnsk, 2017-08-19:
startup, version: 3126 osx x64 channel: stable
executable: /Applications/Sublime Text.app/Contents/MacOS/Sublime Text
working dir: /
packages path: /Users/amyslivets/Library/Application Support/Sublime Text 3/Packages
state path: /Users/amyslivets/Library/Application Support/Sublime Text 3/Local
zip path: /Applications/Sublime Text.app/Contents/MacOS/Packages
zip path: /Users/amyslivets/Library/Application Support/Sublime Text 3/Installed Packages
ignored_packages: ["Vintage"]
pre session restore time: 0.28425
using gpu buffer for window
first paint time: 0.351649
startup time: 0.352949
reloading plugin Default.auto_indent_tag
reloading plugin Default.block
reloading plugin Default.comment
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.install_package_control
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.quick_panel
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.settings
reloading plugin Default.show_scope_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin 0_package_control_loader.00-package_control
reloading plugin Package Control.1_reloader
reloading plugin Package Control.2_bootstrap
reloading plugin Package Control.Package Control
reloading plugin Deviot.Deviot
plugins loaded
reloading settings Packages/User/deviot.sublime-settings
Package Control: Skipping automatic upgrade, last run at 2017-08-19 13:37:03, next run at 2017-08-19 14:37:03 or after
Exception in thread Thread-6:
Traceback (most recent call last):
  File "./python3.3/json/decoder.py", line 367, in raw_decode
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "/Users/amyslivets/Library/Application Support/Sublime Text 3/Packages/Deviot/platformio/upload.py", line 68, in start_upload
    if(not self.check_auth_ota()):
  File "/Users/amyslivets/Library/Application Support/Sublime Text 3/Packages/Deviot/libraries/project_check.py", line 260, in check_auth_ota
    platform = self.get_platform()
  File "/Users/amyslivets/Library/Application Support/Sublime Text 3/Packages/Deviot/libraries/preferences_bridge.py", line 109, in get_platform
    boards = boards_file.read_json()
  File "/Users/amyslivets/Library/Application Support/Sublime Text 3/Packages/Deviot/libraries/file.py", line 57, in read_json
    return json.loads(file)
  File "./python3.3/json/__init__.py", line 316, in loads
  File "./python3.3/json/decoder.py", line 351, in decode
  File "./python3.3/json/decoder.py", line 369, in raw_decode
ValueError: No JSON object could be decoded

maybe it happened after  sudo python ./setup.py install
 for ESP8266 esptool

After I have removed and reinstall Python 2.7 - but still no effect

lexxnsk, 2017-08-19:
reinstall from scratch - UnicodeDecodeError: 'ascii' codec can't decode byte 0xd0 in position 35: ordinal not in range(128)

I have russian text in sketch

gepd, 2017-08-19:
Can you please provide me part of the code to replicate the error?
I'm sure it's because the russian text, the code I'll help me to debug

lexxnsk, 2017-08-19:
Thanks for quick reply.
 Here it is

bool light_status = false;     // Создавем переменную, в которой будем хранить статус вкл\выкл светодиодов
unsigned long time_on ;           // Создаем переменную, в которой будет хранится время включения светодиодов
unsigned long time_off ;          // Создаем переменную, в которой будет хранится время выключения светодиодов
unsigned long interval = 5000 ; // Время, в течение которого светодиоды должны быть включены (в миллисекундах)

void setup() {
  Serial.begin(115200);        // Настраиваем скорость последовательного порта для вывода отладочной информации
  Serial.println("Booting");   // Выводим в последовательный порт слово Booting
  
  #define LED 5                // Выход на светодиодную ленту (MOSFET GATE) 
  #define PIR 4                // Вход с датчика движения
  #define TIMEZONE 3		   // Задаем текущую таймзону
  //   #define LED 16          // Светодиод рядом с CP2102
  //   #define LED 2           // Светодиод рядом с антенной

  pinMode(LED, OUTPUT);        // Настраиваем ножку PIN как выход Initialize the LED pin as an output
  pinMode(PIR, INPUT);         // Настраиваем ножку PIR как вход
  digitalWrite (LED, LOW);     // При загрузке светодиодная лента выключена
  attachInterrupt (PIR, led_on, CHANGE); // Включаем прерывание. При смене логического уровня на ножке PIR выполняем функцию led_on()
}

void led_on() {                               // Создаем функцию led_on
	time_on = millis();                         // Записываем текущее значения времени в переменную time_on
	Serial.println("Сработала функция led_on"); // Выводим текст в последовательный порт
	if (light_status == false) {                // Если светодиоды выключены, то делаем то, что в скобках далее
		Serial.println("Включаем светодиоды");    // Выводим текст в последовательный порт
		detachInterrupt (PIR);                    // Выключаем прерывание на время розжига светодиодов в цикле меняем яркость с 0 до 1023
		for (int brightness = 0; brightness <= 1023; brightness++) {
			// Serial.println(brightness); 
			analogWrite (LED, brightness);
			delay(1);
			}
     	light_status = true;
     	Serial.print("light_status = ");
     	Serial.println(light_status);
  }
  attachInterrupt (PIR, led_on, CHANGE);      // Включаем прерывание. При перехоже ножки PIR из 0 в 1 выполняем led_on ()
}

void led_off() {
	Serial.println("Сработала функция led_off");
	Serial.println("Выключаем светодиоды");  
	detachInterrupt (PIR);  // Выключаем прерывание на время розжига светодиодов
	for (int brightness = 1023; brightness >= 0; brightness--) {
		analogWrite (LED, (brightness));
		delay(1);
		}
	light_status = false;
	Serial.print("light_status = ");
	Serial.println(light_status);
	attachInterrupt (PIR, led_on, CHANGE); // Включаем прерывание. При перехоже ножки PIR из 0 в 1 выполняем led_on()
}   
 

void loop() {
  time_off = (millis()-time_on);          // Считаем разницу между текущим временем и временем включения светодиодов
   Serial.println (time_off);             // Выводим текст в последовательный порт
   Serial.println (digitalRead (PIR));    // Выводим в последовательный порт логический уровень на ножке PIR
   
  if (time_off >= interval && light_status == true && digitalRead (PIR) == 0) {   // 
    Serial.println("Выключаем светодиоды из цикла loop ()");                      // Выводим текст в последовательный порт
    led_off (); 
  }
 
}

lexxnsk, 2017-08-19:
Try  again - same error

Exception in thread Thread-16:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "/Users/amyslivets/Library/Application Support/Sublime Text 3/Packages/Deviot/platformio/compile.py", line 49, in start_compilation
    self.make_cpp_temp()
  File "/Users/amyslivets/Library/Application Support/Sublime Text 3/Packages/Deviot/platformio/initialize.py", line 113, in make_cpp_temp
    self.add_arduino_lib(cpp_file)
  File "/Users/amyslivets/Library/Application Support/Sublime Text 3/Packages/Deviot/libraries/preferences_bridge.py", line 302, in add_arduino_lib
    sketch = file.read()
  File "./python3.3/encodings/ascii.py", line 26, in decode
UnicodeDecodeError: 'ascii' codec can't decode byte 0xd0 in position 38: ordinal not in range(128)

It is strange, why it is linked to python 3.3. Is it OK?

Googled this solution, not sure how to implement, because I have 2.7 python installed on my mac
https://stackoverflow.com/questions/2276200/changing-default-encoding-of-python

gepd, 2017-08-19:
Yes, it's ok, Sublime Text works with python3 (an own version), but PlatformIO is executed in python2, so the problem is reaised by Sublime Text when try to run a "command" in this case with popen

I'm doing some tests too. I did already have a report from other user, I've read that solution too, but it didn't help in this case, at least when I checked it

closed 2017-09-12 by gepd