gepd/Deviot #146: error: 'led_on' was not declared in this scope

lexxnsk, 2017-08-19:
Link to function led_on (described below) is recognized as variable....

bool light_status = false;     // Sozdavem peremennuju, v kotoroj budem hranit' status vkl\vykl svetodiodov
unsigned long time_on ;           // Sozdaem peremennuju, v kotoroj budet hranitsja vremja vkljuchenija svetodiodov
unsigned long time_off ;          // Sozdaem peremennuju, v kotoroj budet hranitsja vremja vykljuchenija svetodiodov
unsigned long interval = 5000 ; // Vremja, v techenie kotorogo svetodiody dolzhny byt' vkljucheny (v millisekundah)

void setup() {
  Serial.begin(115200);        // Nastraivaem skorost' posledovatel'nogo porta dlja vyvoda otladochnoj informacii
  Serial.println("Bofffoting");   // Vyvodim v posledovatel'nyj port slovo Booting
  
  #define LED 5                // Vyhod na svetodiodnuju lentu (MOSFET GATE) 
  #define PIR 4                // Vhod s datchika dvizhenija
  #define TIMEZONE 3		   // Zadaem tekushhuju tajmzonu
  //   #define LED 16          // Svetodiod rjadom s CP2102
  //   #define LED 2           // Svetodiod rjadom s antennoj

  pinMode(LED, OUTPUT);        // Nastraivaem nozhku PIN kak vyhod Initialize the LED pin as an output
  pinMode(PIR, INPUT);         // Nastraivaem nozhku PIR kak vhod
  digitalWrite (LED, LOW);     // Pri zagruzke svetodiodnaja lenta vykljuchena
  attachInterrupt (PIR, led_on, CHANGE);
}

void led_on() {                               // Sozdaem funkciju led_on
	time_on = millis();                         // Zapisyvaem tekushhee znachenija vremeni v peremennuju time_on
	Serial.println("Srabotala funkcija led_on"); // Vyvodim tekst v posledovatel'nyj port
	if (light_status == false) {                // Esli svetodiody vykljucheny, to delaem to, chto v skobkah dalee
		Serial.println("Vkljuchaem svetodiody");    // Vyvodim tekst v posledovatel'nyj port
		detachInterrupt (PIR);                    // Vykljuchaem preryvanie na vremja rozzhiga svetodiodov v cikle menjaem jarkost' s 0 do 1023
		for (int brightness = 0; brightness <= 1023; brightness++) {
			// Serial.println(brightness); 
			analogWrite (LED, brightness);
			delay(1);
			}
     	light_status = true;
     	Serial.print("light_status = ");
     	Serial.println(light_status);
  }
  attachInterrupt (PIR, led_on, CHANGE);      // Vkljuchaem preryvanie. Pri perehozhe nozhki PIR iz 0 v 1 vypolnjaem led_on ()
}

gepd, 2017-08-19:
this should be fixed changing the `led_on` function before `setup`. I'm working in a feature to warn about this and an option to automatically fix it, but meanwhile you should do it manually

lexxnsk, 2017-08-19:
It is strange but before I made something (may be installed esptool python) that script was compiled OK. with led_on after setup and with russian symbols... Magic...
But anyway, thanks for idea - it helped.
Also I moved 
#define LED 5                // Vyhod na svetodiodnuju lentu (MOSFET GATE) 
 #define PIR 4  
commands on top before led_on function (to prevent same error)

gepd, 2018-02-11:
this shouldn't be happening anymore, latest releases of platformio solved this issue

closed 2018-02-11 by gepd

zalexzperez, 2018-05-29:
Still happening...

gepd, 2018-06-01:
@zalexzperez if it still happen, do what I recommend above, move the functions before setup or add a prototype for the funtion(s)