gepd/Deviot #148: Dev Compilation being wierd

tablatronix, 2017-08-24:
Hmm so a few things with new build

* .ino not highlighted with syntax
* Ino being converted to cpp during compilation
* Compiler being stupid and dooing out of order declaration checks probably because of the cpp, maybe a compiler flag, i have seen this error with arduino ide when they had to fix this.

tablatronix, 2017-08-24:
#146

tablatronix, 2017-08-24:
Looks like the platformio arduino preprocessor is not working properly.

gepd, 2017-08-24:
I'm currently working in the highlight problems, there has been some changes in ST that has causing crashing the program with the current syntax, I hope to solve that this week.

PlatformIO has been always converting `ino` files to `cpp`, but now I'm working in a experimental function to solve https://github.com/gepd/Deviot/issues/104 because the `ino` files provoke wrong line numbers when there is an error in the code but it not happens when your file is `cpp`. As that is feature too experimental I'll move to a new branch so at least you can compile without any extra problem

The order declaration, is related with the cpp. In the Arduino IDE you can compile the code with declarations in any order, but this is not happening in PlatformIO. That is why some times a code compile without problem in the Arduino IDE and show an error in Deviot. I'm trying to find a solution to this because the PlatformIO core hasn't, besides to manually rename your sketch to .cpp

tablatronix, 2017-08-24:
Ahh yes the arduino ide puts lines in for code error compatibility to match up original ino.

Hmm platformio has an ino to cpp function i linked above. Is that different that the core you are using?

tablatronix, 2017-08-24:
Oh and the compiler errors are awesome! Loving the changes and progress

tablatronix, 2017-08-24:
And to clarify it is not that arduino preprocessor allows declarations anywhere it is that it auto creates prototypes definitions in the right order I believe. And that is what the inotocpp platformio script probably does. But i do not see any of the tool scripts in deviot..

gepd, 2017-08-27:
Thanks for the info, it will be very useful. 

I already removed all changes of this feature from the develop branch, I'll working on it in: https://github.com/gepd/Deviot/tree/feature/wrong_INO_lines

> Hmm platformio has an ino to cpp function i linked above. Is that different that the core you are using?

At this momment it is an incomplete experiment, but now I'll start to create my own ino to cpp, the idea is to get the same results as the arduino IDE. PlatformIO is suggesting to their users avoid use ino files, but I suppose there may be a better solution

tablatronix, 2017-08-27:
yeah it is a bad practice, but arduino is huge, and it does make jumping into coding much easier for beginners, not having to predefine stuff or worrying about the compiler at all. Makes is more like scripting.

gepd, 2017-08-27:
Yes, without talking about the quantity of sketches already coded in that way

tablatronix, 2017-08-27:
So why do you need to roll your own does deviot onlymuse a subset of platformios functionality or tools?
Or are you really having to do custom stuff to get linting working, and other ide features working? I assumed ctags was all you needed.

gepd, 2017-08-27:
I try to fix issues that are not priority for the PlatformIO team. I always try to add functionalities that I need in my workflow. 

This proyect born as an alternative to Arduino IDE, so should be logic to be have full compatibility with that kind of project, without break the normal functionality with PlatformIO ones.

In the case of the INO files, PlatformIO doesn't like that format and they are not working in a solution for the issues related in this matter, especially with wrong lines linter, that is why I started to implement new tools to solve that small problems

tablatronix, 2017-08-27:
I see well that sucks, but makes sense,  I thought they were actively developing with arduino considered. I guess they want platformio way of doing things to be a thing.