gepd/Deviot #149: dev install instructions

tablatronix, 2017-08-25:
Does anyone have any reliable dev install instructions ?

I was getting errors, then in noticed by package repo was pointing to deviot_2, which seems to now be develop branch. So i changed that now when in install deviot I get

```
'"C:\Users\...\Documents\projects\bin\sublimetext\Data\Packages\User\Deviot\penv\Scripts\platformio"' is not recognized as an internal or external command,
operable program or batch file.
```

and my scripts folder is empty...

I seem to always have problems, is it more reliable to just use git ?

tablatronix, 2017-08-25:
So far I have worked out this.

package manager -> add repository
`https://github.com/gepd/Deviot/tree/develop`

package manager -> install or upgrade package, depending on what you are doing
select deviot ( ignore the fact that this will not show your custom repo url!)

click  preferences 
click browse packages
go up a folder to user/deviot and delete it
restart submlime text
platformio will reinstall

gepd, 2017-08-25:
yes, that is the easiest and fastest way to use the develop branch.

Same instructions are posted in: https://github.com/gepd/Deviot/wiki/Setup#developer
including the way to force update to the latest changes made in the remote branch

tablatronix, 2017-08-25:
So what is the difference between upgrade package and upgrade/overwrite packages. I thought the later just did all packages

gepd, 2017-08-25:
If for example; I push a change and you update to get that last change, it will do normally. But if I push a new change 5 or 10 minutes later, and you try with the upgrade option, it will not download that changes, in that case upgrade/overwrite do works. At least in my case, as I work with multiples virtual machines, I got that behavior when I try to update too often

tablatronix, 2017-08-26:
interesting, i have noticed in sublime logs "skipping upgrade until blah"
I wonder if that is some rate limiting that can be overriden

gepd, 2017-08-26:
According to the package control setting file, there is an option:

```javascript
// The minimum frequency in hours in which to check for automatic upgrades,
// setting this to 0 will always check for automatic upgrades
"auto_upgrade_frequency": 1
```

jwhendy, 2017-09-07:
I just went through installing the dev version and *thought* it worked to add the repo and then upgrade, but some various behavior made me less sure. I ended up just removing/re-installing and think that it would be awesome to change the description in the `Package Control -> Install` menu. This is what I get:

[image]

I can infer that `v2017.09.07` looks more like development than `v2.1.1`, but both list github as the source. I also found it tricky to actually get a version out of Deviot. I upgraded PlatformIO, for example, and this printed out:

```
[ Deviot 2.1.1 ] Starting...
09:50:18 Searching for PlatformIO Updates...

You're up-to-date!
PlatformIO 3.4.1 is currently the newest version available.
```

So that message doesn't differentiate the version at all, and is the same as the released version. `Deviot -> About` just sends me to this repo. It would be great to show the version (git hash) somehow.

tablatronix, 2017-09-07:
Yeah I mentioned that above, package control does not reflect a custom repository in the select.

Deviot console shows version
`[ Deviot 2.1.1 ] Starting...`

jwhendy, 2017-09-07:
@tablatronix I'm guessing I get two items because Deviot is submitted to some sort of Package Control repository. Note that the [install isntructions](https://github.com/gepd/Deviot/wiki/Setup) don't require anything special for the release version. Just install Package Control and then install Deviot from it.

Since I've added the `develop` repo as well, Pacakge Control shows both. That's my guess anyway.

I think between the lines, it is showing a custom repo based on the version. One is release nomenclature (`v2.1.1`) and the other is a date, which signified a rolling release to me.

That said, I think they could be clearer for sure.

tablatronix, 2017-09-07:
oh did you install dev manually?

jwhendy, 2017-09-07:
I just followed [the instructions](https://github.com/gepd/Deviot/wiki/Setup#developer).

- add develop repo via Package Control user settings
- Package Control -> Install package -> type `deviot` into box as shown

Do you only see one option when you do that? Keep in mind that I'd *uninstalled* first, so I think it's showing that I can install either the release or developer version.

tablatronix, 2017-09-07:
I add repo as in my post, package control -> add repository -> paste develop repo url, then install

jwhendy, 2017-09-07:
Seems to do the same. My way produced this user config:
```
{
	"bootstrapped": true,
	"in_process_packages":
	[
	],
	"installed_packages":
	[
		"Deviot",
		"Package Control"
	],
	"repositories":
	[
		"https://github.com/gepd/Deviot/tree/develop"
	]
}
```

I just deleted the entire `repositories:` entry and did it your way. I get this:
```
{
	"bootstrapped": true,
	"in_process_packages":
	[
	],
	"installed_packages":
	[
		"Deviot",
		"Package Control"
	],
	"repositories":
	[
		"https://github.com/gepd/Deviot/tree/develop"
	]
}
```

- Current: `Package Control -> Install package`:
[image]

- `Package Control -> Remove package: Deviot`
- `Package Control -> Install package`, type `deviot`:
[image]

No idea, but that's what I get! I'm still not following what you're confused about. You haven't posted any screenshots to compare what's listed, whether you're looking at the install list with the `develop` branch installed already, with Deviot completely removed like me, etc.

gepd, 2017-09-07:
I've check that, and yes, when you install deviot and let the `repository` option with the develop branch, it will show you deviot twice. 

I'm trying to fix this little problem. I know it started when I update the name of the package in the file `d.json` from `Deviot` to `Deviot (Arduino IDE)` I suppose when it's using the master branch, it will get the name of the repository from that file, and when it's using the `develop` it won't

tablatronix, 2017-09-09:
now what is going on ?
deviot (arduino IDE) is now the package name ?
Is it gonna stay like that, how do i fix, wipe the whole thing , it keeps reinstalling and saying syntac not found

jwhendy, 2017-09-09:
@tablatronix Interesting. I got some notice about a failure to load a file (*something* like `sublime.deviot.settings`, bu don't recall exactly??) and thought it was just me.

Maybe due to how versions were updated this caused a momentary blip for everyone? Reinstalling worked for me as well, even though I think it complained on startup after the change, after removal, and after restarting). After re-install/reboot, it did not complain again.

tablatronix, 2017-09-09:
Yeah my menu was all greyed out aside from a few things ill have to turn off auto updates for now before dev becomes unstable with so much activity

gepd, 2017-09-10:
I've updated the instructions: https://github.com/gepd/Deviot/wiki/Setup#developer

I think now it's easier and it solves the problem with the duplicate entries in the Package Control list

jwhendy, 2017-09-10:
Can you post a full version of what hierarchy this should have in the full config? I'm having a bunch of errors, but likely due to assuming I'd pass this as a completely separate top level `dict`. Probably wrong...

I can't seem to get `develop` to actually install either.  I have this currently (a guess how it should be):

```
{
	"bootstrapped": true,
	"in_process_packages":
	[
	],
	"installed_packages":
	[
		"Deviot (Arduino IDE)",
		"Package Control"
	],
	"packages":
	[
		{
			"details": "https://github.com/gepd/Deviot/tree/develop",
			"donate": "https://gratipay.com/~gepd/",
			"labels":
			[
				"build system",
				"ide",
				"arduino",
				"iot",
				"platformio"
			],
			"name": "Deviot (Arduino IDE)",
			"previous_names":
			[
				"Deviot"
			],
			"releases":
			[
				{
					"branch": "develop",
					"sublime_text": ">=3000"
				}
			]
		}
	],
	"schema_version": "3.0.0"
}
```

Is that correct?

When I try to install, I see only one listing for `Deviot (Arduino IDE)`. It shows the install page, but I have no `Deviot` menu, even after restart. How are the new instructions be pointing to the `develop` branch?

gepd, 2017-09-10:
Nop, when you go to `Preferences > Package Package Control > Add Repository` it will prompt for an url,  paste `https://rawgit.com/gepd/Deviot/master/repository.json` and override the packages

jwhendy, 2017-09-11:
Whoops. I missed that it was `Add Repository` and not edit settings. I saw `paste` and then didn't read on :(

I must have goofed something as I don't have a `Deviot` menu anymore, even though it's installed. I also only get prompted for 2.1.1 install.

gepd, 2017-09-11:
Try using `Upgrade/Override All Packages` 
if it doesn't help, let me know if there is any error in the ST console

jwhendy, 2017-09-11:
I blew away a bunch of stuff in `.config/sublime-text-3` the other day when trying to figure out the complaint about the missing `Deviot` syntax file. I think I messed up ST3 more than expected.

I just reinstalled and things appear to be working.

Did you ditch the idea of `x.y.z-dev.n` for versioning? `develop` branch also just has `2.1.5`, or so it looks that way from `__init__.py`. Or is it just that `develop` and `master` are currently even?

gepd, 2017-09-11:
I've just released the version `2.1.5` to fix all problems related with the package renaming, it had broked the master branch, so it was necessary to do it now

closed 2017-09-19 by gepd