gepd/Deviot #150: DEV extra library asset not found on compile

tablatronix, 2017-08-26:
Before report an issue, please make sure it hasn't been reported before.

### Operating system
OSX

### Board used
nodemcu v1
### Description of problem
add extra library folder

### Steps to Reproduce
add library to ino
#include library

### Actual Results
/Users/.../projects/microcontrollers/dev/libraries/WiFiManager/examples/AutoConnectWithTimeout/AutoConnectWithTimeout.ino:1:25: fatal error: WiFiManager.h: No such file or directory
#include <WiFiManager.h>

### Expected Results
compile

### Additional info
no debug info available

gepd, 2017-08-27:
fixed with: https://github.com/gepd/Deviot/commit/0a0484e8a31be8b5087d51ad0b6d7d0008cf45ac

gepd, 2017-08-28:
Open this issue again if the problem continue

closed 2017-08-28 by gepd